shivasiddharth/GassistPi #100: new

lidense, 2017-12-23:
Hi Shivasiddharth,
I cannot install gassistpi with the new procedure, when I enter the authorization code I get this error message, I tried to register the device manually but it doesn't help, although I have registered the device model, i cannot create an instance (no permission??)
Thanks
(I have put xxx to cover the project number)

File "/home/pi/env/lib/python3.5/site-packages/googlesamples/assistant/library/hotword.py", line 84, in register_device
    raise Exception('failed to register device: ' + r.text)
Exception: failed to register device: {
  "error": {
    "code": 403,
    "message": "Could not create the device instance. You don't have permission to get device model 8054xxxx-20171223xxxx. Make sure you registered this device model under your project 8054xxxx.",
    "status": "PERMISSION_DENIED"

shivasiddharth, 2017-12-23:
I guess your project-id is wrong. The project id will have a name and number. I have shown here as to how you need to copy your project-id. https://youtu.be/vprJBDDYE8Q?t=380

shivasiddharth, 2018-01-03:
Since, i have not heard back, i am assuming that the issue has been resolved. I am closing this issue for now.

closed 2018-01-03 by shivasiddharth

lidense, 2018-01-04:
I actually succeeded in having it work.
But I tested it so much during the day that at the end I was receiving an error message for 24 hours... Does that mean that google can limit access to their APIs as they like when one uses it too much??
That's a pain isn't it?
Thanks and regards
David

----- Mail d'origine -----

https://github.com/shivasiddharth/GassistPi/issues/100#issuecomment-355049963

shivasiddharth, 2018-01-04:
You are allowed 500 free requests per day. In all these months of development, I have hit that mark only once. Thats infact sufficient. People in the enthusiasm of testing may hit the quota initially, but once that enthusiasm wears out, 500 is plenty. Assuming that a person sleeps for 4 hrs and does not mumble wakewords in sleep, the quota translates to about 25/hr i.e 1 request every 2.5 mins (approx). I guess that is sufficient.

lidense, 2018-01-04:
OK thanks and this is free of charge correct?
Is it possible that google might reduce this significantly or that one might have to pay for this in the future?
Thanks and regards
David