shivasiddharth/GassistPi #102: Change language

pmorefr, 2017-12-24:
Hi,

Thank you for sharing your project! It's one of the most complete I've seen to get started with so much addition :)
It's working fine for me, however, since last google update, we should be able to change language. 

And as English is not my native language, I tried to change the lang flag in "action.py". Seems ot enough or not working for me. 
I also tried in the android Home app. I can see the GassistPi device in Home Control. In More Settings, if I change language, then it's not working too. Still in English.

(I don't know if it's related, but I noted in Home Control, I can't change the Nickname&Room, I get a "cannot update setting.. check connection" but maybe this is normal..)

Do you know (or maybe someone else here) how to change the language setting please?

OT notes if you mind (or do you prefer this in an other issue topic):
- which audio driver script is best to use if using USB mic+I2S dac
-  regarding your recent snowboy changes, do you think this can help you https://danicymru.wordpress.com/2017/12/21/aiy-voice-kit-voice-vad-or-hotword-activation/

Merry Christmas and Happy new year!

Enzoinventor, 2017-12-24:
@pmorefr 
hi, the action.py file as well as hotword.py are not yet structured for language change.
you can try the pushtotalk.py command which works with the language change in this way:
googlesamples-assistant-pushtotalk --device-id 'you personal id' --lang it-IT

Enzoinventor, 2017-12-28:
@shivasiddharth hello, a few hours ago has been issued a correction that allows you to change the language in the hotword but only from the assistant app. If we perform the update as shown here: https://github.com/googlesamples/assistant-sdk-python/issues/146
is there a risk of breaking the GassistPi project?

shivasiddharth, 2017-12-28:
No, upgrade will not break the project. I have made the changes to main.py also now, and have tested the language features. It works.
After upgrading, replace the existing main.py with the ones, currently in the git. You should have language support. Since, the language issue has been resolved, closing this issue now. Please feel free to open in case of necessity.

closed 2017-12-28 by shivasiddharth

Enzoinventor, 2017-12-28:
Hi, so you confirm that anyone who can not install the assistant application is cut off and can not change the language? from what I understand it only works on devices with at least android 6.0?

shivasiddharth, 2017-12-28:
If you cannot install the application, you cannot change the language. That is how google has designed it and that is how this project will work. I am not doing something outside the boundaries of google's design.

Enzoinventor, 2018-01-04:
hi, I followed these instructions:

update the package with pip install google-assistant-sdk[samples] --upgrade
delete the previously register device with googlesamples-assistant-devicetool delete --device DEVICE_ID
run googlesamples-assistant-hotword --project_id PROJECT_ID --device_model_id DEVICE_MODEL
switch the locale in the Google Assistant app

how do I delete the device id created in gassistpi? what is the correct passage?
thank you