shivasiddharth/GassistPi #103: pushbutton.py don't work

Enzoinventor, 2017-12-24:
Hi, I noticed that the pushbutton.py command does not work
(env) pi@raspberrypi:~ $ python3 /home/pi/GassistPi/src/pushbutton.py
INFO:root:Connecting to embeddedassistant.googleapis.com
and then it remains locked there

shivasiddharth, 2017-12-24:
As the title of the file says, its for trigger using a pushbutton connected between GPIO 22 and GRND. I have mentioned that in the introduction part of readme and also have mentioned it in the GPIO allocation table.

closed 2017-12-24 by shivasiddharth

Enzoinventor, 2017-12-24:
I apologize. I'm using the aiy kit and the button is connected to the 23 just enough that I edit the right file?

Enzoinventor, 2017-12-24:
ok now it works but the stream news functions and all the others do not work like never?

Sumeetgk, 2018-01-26:
my button is also not working how did u fixed it ?