shivasiddharth/GassistPi #104: Can't Play Music from Youtube

valascus, 2017-12-24:
Well... the title says it all...
Only says "Playing: }". Guess it doesn't get the song's name?
But i did say it and google recognized it.
https://image.prntscr.com/image/GFc8W_vLROKWUeeQNGFCqg.png

Even got a Youtube API key, and i was not sure if it was needed for this, previously, it wasn't.

No errors aparently. Any clue?

valascus, 2017-12-24:
Yes, changed the work to Youtube because i say Stream and google recognzies Dim or 3... Jeez. I'm pretty sure i say it right... So as "tune". Such a hard word to google get it.

valascus, 2017-12-24:
And give me your contact, i can help getting respeaker working :D

shivasiddharth, 2017-12-24:
If you change the keyword in main.py, you need to change it in actions.py also to get the song name printed. in the actions.py, scroll to the function `def YouTube(phrase):` and under there, change the stream to whatever you set the main.py to. 
I am not in need of respeaker, but if you can post it in this issue #89  as to what you did for the respeaker, then it will help the community.

closed 2017-12-24 by shivasiddharth