shivasiddharth/GassistPi #105: pushbutton.py

Enzoinventor, 2017-12-24:
it works but does not perform the stream, news, and all the other functions.
why?

shivasiddharth, 2017-12-24:
Please make sure to go though the keywords and the examples i have given in readme document. If even after using the appropriate keywords, the custom action does not work, then manually start the pushbuton.py and post the error that you see on the terminal. 
If you report an error and then ask me as to why the error is occurring, then i may be in a position to guide you, but just saying it does not work without providing any info does not help the cause and I cannot make some wild guess.
AIY Kit users wanted to use the pushbutton to stop the streams, feeds etc. So i purposed the GPIO23 for closing any related activity (and again, this info is given in the readme document). Now, if you are using that GPIO23 for starting the pushtotalk, then it could have an impact. If you want to change that, then you need to change the GPIO number in stopbutton.service and also in actions.py file in places where i have used it.

Enzoinventor, 2017-12-25:
@shivasiddharth Merry Christmas. sorry for not being well explained actually I forgot to specify that when I pushbutton, on the terminal I read correctly that the stream command was given but the assistant always responds (sorry I do not know how to help) to any function I give answers the same way.

shivasiddharth, 2017-12-25:
@Enzoinventor , Merry Christmas.. I had made some tiny changes to the file. https://github.com/shivasiddharth/GassistPi/blob/master/src/pushbutton.py down this and replace the existing file and try.. Let me know as to how it goes.

Enzoinventor, 2017-12-25:
I tried but still nothing

Enzoinventor, 2017-12-25:
here's what happens:
INFO:root:Recording audio request.
Playing: "
stability: 0.009999999776482582
]

shivasiddharth, 2017-12-25:
I have modified the codes and have tested the music streaming.. here is the update file. https://github.com/shivasiddharth/GassistPi/blob/master/src/pushbutton.py 
The issue with the new gRPC API is that, it is picking up words as the user pronounces them, So as soon as you say stream, it starts processing, without waiting for other words. As a result, you should say the keyword last. For example, with main.py, it will be "Stream I got you", but with pushbutton.py it will be "I got you stream"

Enzoinventor, 2017-12-25:
thank you so much man, the stream now works. but if I say: i get you stream on kody the normal play of youtube starts. there is still some corrections to the code but we are on track.
thank you for your time

shivasiddharth, 2017-12-25:
For Youtube on Kodi, you have got the wrong command.  Is the command that you used is what i have given in that big table ? I again repeat, please go through the command syntax that i have given.  By default, both in main.py and pushbutton.py, "On Kodi" is pronounced last, so there is no change needed.

shivasiddharth, 2017-12-26:
Since your issue was resolved, i am closing this for now, if anything else comes up please feel free to open it again.

closed 2017-12-26 by shivasiddharth

Enzoinventor, 2017-12-26:
as I said, the code is to be corrected because if I ask the assistant what you are doing, he should listen again after speaking and instead he will stop.
even when the assistant answers (what can I do for you?) he should listen and instead stop himself.
in the original pushtotalk example in googlesamples it works so you should correct it.
thank you so much

shivasiddharth, 2017-12-26:
Ok, you mean the conversation is not continuing, i have just fixed it. Replace you existing file with this. https://github.com/shivasiddharth/GassistPi/blob/master/src/pushbutton.py Thanks for pointing it out. But only if your precise about the issue, i can do something. If you just say "some corrections" it really does not give me any idea as to what you are talking about.

Enzoinventor, 2017-12-26:
I fully agreed that the conversation did not continue. I immediately try your correction and let you know. for now I thank you very much.

Enzoinventor, 2017-12-26:
ok is back listening, great job.
I encountered this new problem when pronouncing words with the accent like (cos'è)= what is  in Italian
[transcript: "cos \ '\ 303 \ 250"

shivasiddharth, 2017-12-26:
This is not an error, terminal as it is cannot print special characters.

Enzoinventor, 2017-12-26:
ok grazie mille

Enzoinventor, 2017-12-28:
hi, I found another error in the feed section. it only works topnews, if instead I say sport news I get this error:
INFO:root:Transcript of user request: "Sport News".
[transcript: "Sport News"
stability: 0.009999999776482582
]
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/pushbutton.py", line 539, in <module>
    main()
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/GassistPi/src/pushbutton.py", line 527, in main
    continue_conversation = assistant.assist()
  File "/home/pi/env/lib/python3.5/site-packages/tenacity/__init__.py", line 214, in wrapped_f
    return self.call(f, *args, **kw)
  File "/home/pi/env/lib/python3.5/site-packages/tenacity/__init__.py", line 295, in call
    start_time=start_time)
  File "/home/pi/env/lib/python3.5/site-packages/tenacity/__init__.py", line 252, in iter
    return fut.result()
  File "/usr/lib/python3.5/concurrent/futures/_base.py", line 398, in result
    return self.__get_result()
  File "/usr/lib/python3.5/concurrent/futures/_base.py", line 357, in __get_result
    raise self._exception
  File "/home/pi/env/lib/python3.5/site-packages/tenacity/__init__.py", line 298, in call
    result = fn(*args, **kwargs)
  File "/home/pi/GassistPi/src/pushbutton.py", line 226, in assist
    feed(str(usrcmd).lower())
  File "/home/pi/GassistPi/src/actions.py", line 198, in feed
    feed=feedparser.parse(URL)
UnboundLocalError: local variable 'URL' referenced before assignment