shivasiddharth/GassistPi #109: demo not running

ErkFrr, 2017-12-26:
i receive this error:

pi@raspberrypi:~/env/bin $ ./google-assistant-demo
usage: google-assistant-demo [-h] [--credentials OAUTH2_CREDENTIALS_FILE]
                             --device_model_id DEVICE_MODEL_ID
                             [--project_id PROJECT_ID]
google-assistant-demo: error: the following arguments are required: --device_model_id

any ideas?

shivasiddharth, 2017-12-26:
The demo will auto start on installation, you need not manually run it. After the SDK update, you need to mention the model id and project id for the assistant to work. The new demo code is:
`googlesamples-assistant-hotword --project_id $projid --device_model_id $modelid`

I have explained the installation in detail in this video: https://youtu.be/vprJBDDYE8Q

ErkFrr, 2017-12-26:
Thanks for the fast reply, i Unfortunately never get prompted for the 
project id after i put in my credentials it just kicks me back to the 
bash prompt.

shivasiddharth, 2017-12-27:
At the start of the installation, you will be prompted to enter your credentials file name and your project id did you get that  ? And after you are through the installation and after you authenticate, the demo will auto start.

shivasiddharth, 2018-01-07:
Since, i did not hear back, i am assuming that the issue has been resolved and so, i am closing this. Please feel free to reopen this issue if it has not been resolved yet.

closed 2018-01-07 by shivasiddharth

ErkFrr, 2018-01-07:
Yes I got it running smoothly, thanks again for your speedy replies..  now
I am just trying to get snowboy to work along side it, proving difficult
sofar, I am sure I'll figure it out eventually.

ErkFrr, 2018-01-09:
howdy!

I wanted to thank you for your good work on the google assistant for the 
raspberry pi.  as per usual, thanks comes with a request, so ive been 
messing with the code you put together and i have replaced your radio 
for pianobar(command line pandora) streaming for music, and i have the 
ability to start the stream and kill the stream, but id like to figure 
out how to manipulate pianobar with my voice, regarding skipping tracks 
and the like;

in actions.py i replaced your radio section with this code

#Radio Station Streaming
def radio(phrase):
             say("lets rock out")
             p = 
subprocess.Popen(["pianobar"],stdin=subprocess.PIPE,stdout=subprocess.PIPE)
#kill Radio Station Streaming
def kill(phrase)
             p = 
subprocess.Popen(["killall","pianobar"],stdin=subprocess.PIPE,stdout=subprocess.PIPE)

any advice you would be willing to part with on the matter would be 
greatly appreciated,

thanks again for your time

shivasiddharth, 2018-01-09:
I am hearing about pianobar for the first time, if you could share some links for pianobar control syntaxes, it will be helpful.
Regarding your code, there is no need to define a separate function for terminating pianobar process (unless you are listening to a mixture of audio streams). You can add that line of code under stop function that i have already defined..

ErkFrr, 2018-01-09:
Thanks again for replying so promptly..
https://github.com/PromyLOPh/pianobar
There is a link to the pianobar git.

cheflow2, 2018-01-11:
hi i am getting this problem too 

> google-assistant-demo [-h] [--credentials OAUTH2_CREDENTIALS_FILE]
                             --device_model_id DEVICE_MODEL_ID
                             [--project_id PROJECT_ID]
google-assistant-demo: error: the following arguments are required: --device_model_id

>

ifb777, 2018-01-17:
Hello
I am experiencing this error running the script also'

`google-assistant-demo: error: the following arguments are required: --device_model_id`

What is the device_model_id and where can it be found?

I can't progress beyond this requirement at the moment so further time spent on this would be wasted.
Any input from the collective wisdom of the group would be most welcome.
TIA

ErkFrr, 2018-01-17:
I scrapped and started fresh, and really reading the tutorial, and it
worked fine

shivasiddharth, 2018-01-17:
@ifb777 , google-assistant-demo command does not work after the 20th Dec SDK update and there is no need for this command ass well. Please go through the Git Readme.

ifb777, 2018-01-18:
Thanks for the prompt reply. Yep, should have read the "Read Me" the clue to success was in there.
All up and running now and impressive it is too.  Please keep working on this.  I'm interested to see where this goes.

Thanks again