shivasiddharth/GassistPi #110: Wakeword doesn't seem to work after reboot

seremana, 2017-12-26:
Hello,

First of all, thank you for this nice consolidation of google assistant PI tools.
Installation went successfull, however, the wakeword doesn't seem to work after PI reboot.  I hear the welcome sound after restart.

The assistent works smooth after manual start (python -u /home/pi/GassistPi/src/main.py --project_id 'xyz'  --device_model_id 'xyz-n')

Any idea what can be going on?

Thank you!

shivasiddharth, 2017-12-26:
Did you set the model id and project id in the systemd file ? Please see point:1 under [headless-autostart-on-boot-service-setup](https://github.com/shivasiddharth/GassistPi#headless-autostart-on-boot-service-setup)

seremana, 2017-12-26:
I did, and it does work after trying some time. Just not that good, it appears to have difficulties understanding me.  Could be my mic, could be software related.  I'll look into it.  Thanks!

shivasiddharth, 2017-12-31:
Since the issue seems to be resolved, closing this issue !

closed 2017-12-31 by shivasiddharth