shivasiddharth/GassistPi #114: unable to use gpio triggering

amanmathurastro, 2017-12-28:
i have install modified version of google assistant sdk but not able to use the gpio pins using hey google its not responding .......??? help me out as soon as possible

shivasiddharth, 2017-12-29:
Did you add the project-id and model-id ? Only if you can share the error, i can comment on that.

amanmathurastro, 2017-12-29:
yes i did that ..... actually it is responding in every query but when i say to on the kitchen lights or trigger servo 90 it didn't respond. It also not responding to any gpio related query.

shivasiddharth, 2017-12-29:
It may not be decoding your query properly. stop the assistant using:  
`sudo systemctl stop gassistpi-ok-google.service`
Start the main.py manually using:
` /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id 'project-id'  --device_model_id 'saved-model-id'`
Now give your command and see whether its detecting your command.

amanmathurastro, 2017-12-29:
thank you sir.... just last quetion than i"ll close the issue instead of this can i use node mcu one to serve my purpose?

amanmathurastro, 2017-12-29:
thank you sir .... i am closing this issue

closed 2017-12-29 by amanmathurastro