shivasiddharth/GassistPi #115: vlc does not open through the google assisstant

Baidya99, 2017-12-28:
I made a function in actions.py which makes VLC play all the songs in Music folder.
After hours of fiddling I found out that the same command was running VLC when used in a different python file.
Due to VLC not working the radio does not play.
I think the problem is that VLC cannot run as root which is required when starting a service. This is because when I tried running the above mentioned file through a .service it gave me error saying VLC cannot run as root.
Running the python file through actions.py does not work(open vlc) either.

shivasiddharth, 2017-12-29:
VLC does work through the assistant. Otherwise i would not have used it in the project. If you want me to take a look into your code, share them here.

Baidya99, 2017-12-29:
My code:
`def music():
   musp = subprocess.Popen(["/usr/bin/vlc","/home/pi/Music],stdin=subprocess.PIPE,stdout=subprocess.PIPE)
   say("Playing all songs")`

When I ask it to 'play all songs'(which is the keyword used') it responds by saying "playing all songs" but it doesn't play them.
Same happen when I ask it to 'tune into radio <x>'.

Baidya99, 2017-12-29:
It is working now.

closed 2017-12-29 by Baidya99