shivasiddharth/GassistPi #116: Error after saving credentials

zerocolTHS, 2017-12-30:
After running " sudo  /home/pi/GassistPi/scripts/gassist-installer-pi-zero.sh " 

Everything installs fine but then i get this. 

credentials saved: /home/pi/.config/google-oauthlib-tool/credentials.json
Usage: googlesamples-assistant-devicetool register-model [OPTIONS]

Error: Missing option "--type".  Choose from LIGHT, SWITCH, OUTLET.

Im using a pizero.

shivasiddharth, 2017-12-30:
Did you modify the script ? Because straight-away it will work without issues, because i have declared the type in script.

zerocolTHS, 2017-12-30:
Nope i have not modified anything in the script.

shivasiddharth, 2017-12-30:
Open a terminal, move into the environment using:
`source env/bin/activate`
and then execute:
```
googlesamples-assistant-devicetool register-model --manufacturer "Pi Foundation" \
          --product-name $prodname --type LIGHT --model $modelid
```
make sure to change $prodname (roduct name) and $modelid (model-id). You can find your model-id in a text file in home/pi/ directory.

zerocolTHS, 2017-12-30:
(env) 
$ googlesamples-assistant-devicetool register-model --manufacturer "Pi Foundation" \ --product-name $prodname --type LIGHT --model $modelid
Error: --model option requires an argument

shivasiddharth, 2017-12-30:
That is what i asked you to do, change the product name and model-id and not just paste the code. You can find your model-id in a text file in home/pi/ directory.

zerocolTHS, 2017-12-30:
googlesamples-assistant-devicetool register-model --manufacturer "Pi Foundation" \ --product-name REMOVED --type LIGHT --model REMOVED

Usage: googlesamples-assistant-devicetool register-model [OPTIONS]

Error: Missing option "--product-name".

shivasiddharth, 2017-12-30:
What product name did you use ?

zerocolTHS, 2017-12-30:
rock-objective-167517

shivasiddharth, 2017-12-30:
You cannot use '-', numbers in product name. Here is some material for reference https://developers.google.com/assistant/sdk/reference/device-registration/device-tool 
Try "Test Lamp" . Since the name has space, you need to have them within quotes.

zerocolTHS, 2017-12-30:
Well that's what i got from the text file called modelid.txt in home/pi/ as you mentioned. 
SO how would i fix it ?

shivasiddharth, 2017-12-30:
What you have in the text file is called model-id not product name. Please read through the comments carefully. I said, that model-id will be in the text file.

zerocolTHS, 2017-12-30:
I give up, i'm going to reflash and start again.
Thanks for the help you might as well close this.

closed 2017-12-30 by shivasiddharth