shivasiddharth/GassistPi #117: Installation process failing

pskmoorhead, 2017-12-30:
Hi - I'm following the current process on GassistPi -- Google Assistant for all Raspberry Pi Boards exactly using latest Raspbian download.
I'm using the AIY HAT.
The /home/pi/GassistPi/scripts/snowboy-deps-installer.sh script fails in two places. 
firstly building the wheel for pyaudio.  That can be overcome by first running
sudo apt install libasound-dev portaudio19-dev libportaudio2 libportaudiocpp0 ffmpeg libav-tools 

the second fail is on building the wheel for GPIO.
error: invalid command 'bdist_wheel'
the script proceeds to the next step setup of GPIO but it hangs forever on the spinner.
I can't figure out the problem.
any suggestions?

shivasiddharth, 2017-12-30:
1. Which branch are you using? Because i dont have the snowboy dependencies script now. 
2.  Right after the build failure message, right below it, you can see the successfully installed message. Not sure as to why people fail to notice that.

pskmoorhead, 2017-12-30:
HI – thanks for quick reply.

I’m starting from this page
https://github.com/shivasiddharth/GassistPi
and following the instructions for Pi 3 plus AIY-HAT

On running the gassist installer script I get as far as the output below (having previously fixed the problem with the pyaudio wheel build failure).

Requirement already up-to-date: setuptools in ./env/lib/python3.5/site-packages (from protobuf>=3.0.0->googleapis-common-protos>=1.5.2->google-assistant-grpc==0.1.0; extra == "samples"->google-assistant-sdk[samples])
Building wheels for collected packages: grpcio
  Running setup.py bdist_wheel for grpcio ... error
  Complete output from command /home/pi/env/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-mk_bprtk/grpcio/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmpr1ha7ga9pip-wheel- --python-tag cp35:
  Found cython-generated files...
  usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
     or: -c cmd --help

  error: invalid command 'bdist_wheel'

  ----------------------------------------
  Failed building wheel for grpcio
  Running setup.py clean for grpcio
Failed to build grpcio
Installing collected packages: grpcio, google-assistant-grpc
  Running setup.py install for grpcio ... \

At this point the device sits forever (an hour at least) without completing this step – the spinner continues to rotate very slowly

Thanks,
                Paul

shivasiddharth, 2017-12-30:
The 'install for grpcio' will take a lot of time. Thats normal.

pskmoorhead, 2017-12-30:
Ok. Does that mean that the previous error can be ignored?

Building wheels for collected packages: grpcio
  Running setup.py bdist_wheel for grpcio ... error
  Complete output from command /home/pi/env/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-mk_bprtk/grpcio/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmpr1ha7ga9pip-wheel- --python-tag cp35:
  Found cython-generated files...
  usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
     or: -c cmd --help

  error: invalid command 'bdist_wheel'

  ----------------------------------------
  Failed building wheel for grpcio
  Running setup.py clean for grpcio
Failed to build grpcio

shivasiddharth, 2017-12-30:
You can ignore those errors.

pskmoorhead, 2017-12-30:
Thank you sir – it eventually completed the process and after a reboot I seem to be up and running.

Is there a plan to re-introduce custom hotwords again at some stage?

shivasiddharth, 2017-12-30:
Will do it probably sometime later. But, not in the coming weeks though.

sugoei, 2017-12-31:
I had similar errors and stoped installation. so, I couldn't see the Google assistent Demo.
`Failed building wheel for pyaudio
  Running setup.py clean for pyaudio
Failed to build pyaudio
Installing collected packages: pyaudio
  Running setup.py install for pyaudio ... done
Successfully installed pyaudio-0.2.11
Collecting aftership
  Downloading https://www.piwheels.hostedpi.com/simple/aftership/aftership-0.2-py3-none-any.whl
Collecting python-dateutil (from aftership)
  Downloading python_dateutil-2.6.1-py2.py3-none-any.whl (194kB)
    100% |████████████████████████████████| 194kB 405kB/s 
Collecting requests (from aftership)
  Downloading requests-2.18.4-py2.py3-none-any.whl (88kB)
    100% |████████████████████████████████| 92kB 635kB/s 
Collecting six>=1.5 (from python-dateutil->aftership)
  Downloading six-1.11.0-py2.py3-none-any.whl
Collecting urllib3<1.23,>=1.21.1 (from requests->aftership)
  Downloading urllib3-1.22-py2.py3-none-any.whl (132kB)
    100% |████████████████████████████████| 133kB 546kB/s 
Collecting idna<2.7,>=2.5 (from requests->aftership)
Exception:
Traceback (most recent call last):
  File "/home/pi/env/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py", line 594, in urlopen
    chunked=chunked)
  File "/home/pi/env/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py", line 391, in _make_request
    six.raise_from(e, None)
  File "<string>", line 2, in raise_from
  File "/home/pi/env/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py", line 387, in _make_request
    httplib_response = conn.getresponse()
  File "/usr/lib/python3.5/http/client.py", line 1198, in getresponse
    response.begin()
  File "/usr/lib/python3.5/http/client.py", line 297, in begin
    version, status, reason = self._read_status()
  File "/usr/lib/python3.5/http/client.py", line 266, in _read_status
    raise RemoteDisconnected("Remote end closed connection without"
http.client.RemoteDisconnected: Remote end closed connection without response

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/home/pi/env/lib/python3.5/site-packages/pip/commands/install.py", line 353, in run
    wb.build(autobuilding=True)
  File "/home/pi/env/lib/python3.5/site-packages/pip/wheel.py", line 749, in build
    self.requirement_set.prepare_files(self.finder)
  File "/home/pi/env/lib/python3.5/site-packages/pip/req/req_set.py", line 380, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/home/pi/env/lib/python3.5/site-packages/pip/req/req_set.py", line 554, in _prepare_file
    require_hashes
  File "/home/pi/env/lib/python3.5/site-packages/pip/req/req_install.py", line 278, in populate_link
    self.link = finder.find_requirement(self, upgrade)
  File "/home/pi/env/lib/python3.5/site-packages/pip/index.py", line 465, in find_requirement
    all_candidates = self.find_all_candidates(req.name)
  File "/home/pi/env/lib/python3.5/site-packages/pip/index.py", line 423, in find_all_candidates
    for page in self._get_pages(url_locations, project_name):
  File "/home/pi/env/lib/python3.5/site-packages/pip/index.py", line 568, in _get_pages
    page = self._get_page(location)
  File "/home/pi/env/lib/python3.5/site-packages/pip/index.py", line 683, in _get_page
    return HTMLPage.get_page(link, session=self.session)
  File "/home/pi/env/lib/python3.5/site-packages/pip/index.py", line 792, in get_page
    "Cache-Control": "max-age=600",
  File "/home/pi/env/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py", line 501, in get
    return self.request('GET', url, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/pip/download.py", line 386, in request
    return super(PipSession, self).request(method, url, *args, **kwargs)
  File "/home/pi/env/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py", line 488, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/pi/env/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py", line 609, in send
    r = adapter.send(request, **kwargs)
  File "/home/pi/env/share/python-wheels/CacheControl-0.11.7-py2.py3-none-any.whl/cachecontrol/adapter.py", line 47, in send
    resp = super(CacheControlAdapter, self).send(request, **kw)
  File "/home/pi/env/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/adapters.py", line 423, in send
    timeout=timeout
  File "/home/pi/env/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py", line 643, in urlopen
    _stacktrace=sys.exc_info()[2])
  File "/home/pi/env/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/util/retry.py", line 315, in increment
    total -= 1
TypeError: unsupported operand type(s) for -=: 'Retry' and 'int'
pi@raspberrypi:~ $ `

pskmoorhead, 2017-12-31:
run this command:
sudo apt install libasound-dev portaudio19-dev libportaudio2 libportaudiocpp0 ffmpeg libav-tools

then rerun the gaassist installation script
ignore the failure message for gpio wheel
allow the setup.py for gpio to complete (takes a loooong time)
and then wait for script to finish.
Then follow steps to automate headless operation and reboot
worked for me.

shivasiddharth, 2017-12-31:
@sugoei "TypeError: unsupported operand type(s) for -=: 'Retry' and 'int'" is caused by temporary network glitches. if you retry the installation, it will work

sugoei, 2018-01-01:
Thanks both of you. but, it doesn't work. Still, I have python-wheels error. 
Also, I had tested google assistant SDK in google homepage and I met a error.

$ env/bin/python -m pip install --upgrade pip setuptools
same error as above.

shivasiddharth, 2018-01-01:
@sugoei , Is the setup terminating with this "TypeError: unsupported operand type(s) for -=: 'Retry' and 'int'" error?  or do just see the python wheels error in-between the setup (which is normal, as i have already explained above)

sugoei, 2018-01-01:
@shivasiddharth, setup terminated with this error. "TypeError: unsupported operand type(s) for -=: 'Retry' and 'int'"

shivasiddharth, 2018-01-09:
Whats the status of the error ? As i mentioned before, 'Retry' and 'int'" is a network issue. It should have got resolved if you tried later.

sugoei, 2018-01-11:
I had met exception error when install pi3.sh, and installer stopped installation. So, I tried one more time and met this.
mv:cannot stat '/home/pi/GassistPi/src/end.lua'"no such file or directory.
I really want to see google authentification process...

shivasiddharth, 2018-01-13:
If you are running the script for the second time, then comment the mkdir and mv lines in the installation script.

sugoei, 2018-01-14:
Thank you. It's working!!!!!

shivasiddharth, 2018-01-14:
Closing this issue then !

closed 2018-01-14 by shivasiddharth