shivasiddharth/GassistPi #118: Error after Authorisation Code

coconuthero, 2017-12-30:
Hello Sid,
I am new to raspberri (and Linux) but I have followed your procedure to set up the google Assistant very closely. Unfortunately I get this error message after inserting my authorization code and I don't know how to go on from here...Until this point everything went quite well...any suggestion how to proceed from here?

credentials saved: /home/pi/.config/google-oauthlib-tool/credentials.json
Creating new device model
Error: Failed to register model: 400
Could not create the device model. Check that the request contains the required field device_model_idwith a valid format in the request payload. See https://developers.google.com/assistant/sdk/reference/device-registration/model-and-instance-schemas for more information.
pi@raspberrypi:~ $

shivasiddharth, 2017-12-30:
If you can provide answers to these questions, it will be helpful.
1. What OS image are you using? Standard Raspbian OS or the AIY image ?
2. Which Pi are you using?
3. Did you by any chance rename the credentials file after downloading ?
4. What was the product name and project id that you entered when you were prompted for ?

coconuthero, 2017-12-30:
Thank you for your quick reply!

1. Standart Raspian Stretch with Desktop (Nov 2017) Downloaded here: https://www.raspberrypi.org/downloads/raspbian/
2. Raspberry Pi 3 Model B
3. No. 
4. Product Name: GoAssi
Project ID: gassi-190617

shivasiddharth, 2017-12-30:
Could have been due to some temporary network issue. Open a terminal and type the following:
`source env/bin/activate`
After that, the following, make sure to replace $prodname with the product name that you mentioned and $modelid with the modelid given in the text file, which you can find in /home/pi/  directory:
```
googlesamples-assistant-devicetool register-model --manufacturer "Pi Foundation"  --product-name $prodname --type LIGHT --model $modelid
```
You will get a device registered msg, after that execute the following to run the demo make sure to replace $projid with your project-id and $modelid again with the model-id in that text file :

`googlesamples-assistant-hotword --project_id $projid --device_model_id $modelid
`

coconuthero, 2017-12-30:
Thanks! Unfortunately I forgot to plugin the microphone since I cannot use it together with the keyboard...so I got the error message below....(replaced id`s with xxxx)I plugged it in now and write with the virtual keyboard ...but what to go now?

ALSA lib pcm_direct.c:1605:(_snd_pcm_direct_get_slave_ipc_offset) Invalid value for card
device_model_id: xxxx
device_id: xxxx

https://embeddedassistant.googleapis.com/v1alpha2/projects/xxxx/devices/097BF4D6099EE518624ED35534CDDF3D 404
Registering....[12415:12437:ERROR:audio_input_processor.cc(748)] Input error
Device registered.
ON_MUTED_CHANGED:
  {'is_muted': False}
ON_START_FINISHED
[12415:12439:ERROR:audio_input_processor.cc(748)] Input error

shivasiddharth, 2017-12-30:
Everything is looking fine. For that audio error, 
Open a terminal, type:
`sudo nano ~/.asoundrc`
Open a second terminal and type:
`sudo nano /etc/asound.conf`

Just mke sure that contents of the asoundrc is same as the contents of asound.conf. If not, copy the contents of asound.conf into asoundrc. You have your assistant running, so things are looking fine.

coconuthero, 2017-12-30:
good to hear! yes both filles are the same indeed! But there is no reaction to "ok google"....

shivasiddharth, 2017-12-30:
Open a terminal and type:
`sudo systemctl status gassistpi-ok-google.service `
Paste the contents of that terminal here.

coconuthero, 2017-12-30:
pi@raspberrypi:~ $ sudo systemctl status gassistpi-ok-google.service
Unit gassistpi-ok-google.service could not be found.
pi@raspberrypi:~ $

shivasiddharth, 2017-12-30:
so you have not enabled the headless setup yet ? Fine. Manually run the assistant like you did before and paste the what you are seeing! Since, you plugged in the mic in-between, a restart is advisable.

coconuthero, 2017-12-30:
yes I did not continue with headless because you mentioned in your instructions to verify a working assistant before continuing...
Sorry but I don`t know how to manually run the assistant...

shivasiddharth, 2017-12-30:
open a terminal and type:
`/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id '$project-id'  --device_model_id '$saved-model-id'`
like i mentioned previously, insert your project-id and model-id in appropriate places.

coconuthero, 2017-12-30:
oh....sorry...of course...here is the result:
pi@raspberrypi:~ $ /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id gassi-190617 --device_model_id gassi-190617\-20171230181656
device_model_id: gassi-190617-20171230181656
device_id: 097BF4D6099EE518624ED35534CDDF3D

https://embeddedassistant.googleapis.com/v1alpha2/projects/gassi-190617/devices/097BF4D6099EE518624ED35534CDDF3D 200
ON_MUTED_CHANGED:
  {'is_muted': False}
ON_MUTED_CHANGED:
  {'is_muted': False}
ON_START_FINISHED
ON_START_FINISHED

I got a female voice saying hi...but nothing happens when I say "ok google"...I noticed however that under audio device settings the soundcard keeps defaulting to bcm2835 alsa instead of usb audio device...

shivasiddharth, 2017-12-30:
i dont see any errors at all. What is your audio setup (mic and speaker)? What audio config did you use from here https://github.com/shivasiddharth/GassistPi/blob/master/README.md#install-audio-config-files ? Please do not change anything in audio device settings.

coconuthero, 2017-12-30:
I used the following one since I have a microphone connected to a usb-audio-adapter:

3.4. USB MIC AND AUDIO JACK users,

sudo chmod +x /home/pi/GassistPi/audio-drivers/USB-MIC-JACK/scripts/usb-mic-onboard-jack.sh  
sudo /home/pi/GassistPi/audio-drivers/USB-MIC-JACK/scripts/usb-mic-onboard-jack.sh  
speaker-test

shivasiddharth, 2017-12-30:
The assistant is working fine, and since you confirmed the contents of asoundrc and asound.conf to be same I dont see any issue here. Since you mentioned about plugging in the mic late, a restart is warranted. After restart, try to manually start the assistant, it should work. If it does not work even after that, i would advise you to try another mic. As such i really dont see any issue as it is.

coconuthero, 2017-12-31:
ok thank you very much for your time and patience! I had bought the mic and adapter today but I will look for another one...is there anyway I can compensate for your guidance and help?

shivasiddharth, 2017-12-31:
I put this project together to help the pi community and help new users, so never mind... What is that mic and adapter ? May be i can do some digging to see if that could be the issue. open a terminal and execute:
`arecord -l`
that is a lower case L and not capital i

coconuthero, 2017-12-31:
you are great!

pi@raspberrypi:~ $ arecord -l
**** Liste der Hardware-Geräte (CAPTURE) ****
Karte 1: Device [USB Audio Device], Gerät 0: USB Audio [USB Audio]
  Sub-Geräte: 0/1
  Sub-Gerät #0: subdevice #0
pi@raspberrypi:~ $

coconuthero, 2017-12-31:
I got it! It`s the mic! I had another one laying around and had to do some soldering but now it works!
Thank you so much for your support! 
I will leave the headless stuff for tomorrow...good night!

shivasiddharth, 2017-12-31:
Ok, closing this issue then !

closed 2017-12-31 by shivasiddharth