shivasiddharth/GassistPi #119: Finding google assistant authentication link

amplector, 2018-01-05:
On configuring AFTER setting up audio section, #4. What am I looking for in this terminal text? Any example?

supreme83, 2018-01-05:
Follow the first instruction after audio to obtain credentials that you will download and save to your server. It will create the authentication files to let your server access your google account.

ie  Download credentials(from the link attached....tthat requires you following the steps to set up a google developer account and after following the steps you will download a file named---><PROJECT NAME>.json file (refer to this doc for creating credentials https://developers.google.com/assistant/sdk/develop/python/config-dev-project-and-account)

Place the .json file in/home/pi directory DO NOT RENAME

shivasiddharth, 2018-01-06:
@amplector Take a look at this : https://youtu.be/vprJBDDYE8Q?t=755 this how you will be prompted for authentication. If that is what you meant.

crispylel, 2018-01-11:
@shivasiddharth i accidentally pasted blank line instead of auth code and it showed an error; how do i get back to this part without re-installing the whole thing?

shivasiddharth, 2018-01-13:
execute:
source env/bin/activate
then
google-oauthlib-tool --client-secrets /home/pi/$credname --scope https://www.googleapis.com/auth/assistant-sdk-prototype --save --headless
googlesamples-assistant-devicetool register-model --manufacturer "Pi Foundation" \
          --product-name $prodname --type LIGHT --model $modelid

replace $credname, $prodname and $modelid with credentials file name, a product name and modelid respectively.

shivasiddharth, 2018-01-28:
Since, i have not heard back, I am assuming that the issue has been resolved. So closing this for now.

closed 2018-01-28 by shivasiddharth

iamhollis84, 2018-05-10:
How do I get the authorization code