shivasiddharth/GassistPi #120: Language

coconuthero, 2018-01-05:
Hi Sid,

how can I change the language of the google assistant to german?

Thanks!

shivasiddharth, 2018-01-06:
Open the assistant app on phone/tablet, select more settings, choose your device and change the language.

closed 2018-01-06 by shivasiddharth

coconuthero, 2018-01-06:
I checked the assistant app on my Sony Xperia Z5 with Android 7.1.1 thorougly but the raspberry pi is not listed although the assistant is active on the pi. I also checked the google home app without succsess...

shivasiddharth, 2018-01-10:
i have shown here as to how you can access the settings https://youtu.be/vprJBDDYE8Q?t=999

coconuthero, 2018-01-10:
This seems to be disabled in Switzerland

BadFlo, 2018-01-22:
Same here, cannot see the GAssist RasPi on Google Home app ... in Switzerland.