shivasiddharth/GassistPi #121: Autostart Assistant

coconuthero, 2018-01-06:
Hi Sid,
I want to make a magic mirror. How can I automatically start google assistant after booting?
Thank you!

shivasiddharth, 2018-01-06:
I have clearly given that in the README https://github.com/shivasiddharth/GassistPi/blob/master/README.md#headless-autostart-on-boot-service-setup

closed 2018-01-06 by shivasiddharth

coconuthero, 2018-01-06:
I apologize for my question but doesn't headless Autostart mean, that the
screen will be switched off? (I need the screen for my project) Sorry if I
understand it wrong!

shivasiddharth/GassistPi/blob/master/README.md#headless-
autostart-on-boot-service-setup

shivasiddharth, 2018-01-10:
Headless does not mean that display will not work. Headless means that your code will auto start on boot and run in the background.

coconuthero, 2018-01-10:
That's good news! I googled headless and according to the definition I
found it means that a system is running without graphical interface.