shivasiddharth/GassistPi #122: Can't Use Library based Kodi functions

supreme83, 2018-01-06:
[actions.txt](https://github.com/shivasiddharth/GassistPi/files/1608779/actions.txt)
Testing through build. "OK Google, Play Avengers movie on Kodi" restarts Google Assistant service

 and i hear the start up chime and "Hi" from the assistant. 

I've attached my modified actions.py file and can provide any logs necessary, provided you can tell me how to retrieve them

shivasiddharth, 2018-01-07:
It could be because of the NAS directory that you have given. Stop the assistant service using: `sudo systemctl stop gassistpi-ok-google.service` and manually start the main.py. So now when the assistant crashes, it will throw out the reason for the error.

supreme83, 2018-01-08:
[actions.py.txt](https://github.com/shivasiddharth/GassistPi/files/1611638/actions.py.txt)

Thanks for the reply. I actually corrected the issue by simply mounting the remote location as nfs drives and editing /etc/fstab to mount the locations at boot and adjusted Raspbian to pause and wait for network at boot to allow network discovery and mounting. Works great. Unfortunatley Google AIY was limited usage so i decided to try the https://github.com/shivasiddharth/Assistants-Pi setup. It works great  with both alexa and aiy but is it installing the same directories? I ask because GassistPi did not seem to need to be deleted and works along side the Assistants setup but AIY was constantly triggered by my tv and wasting api quota for aiy, so i first renamed entire GassistPi file to Gassist.backup which made both Alexa and AIY stop working on next boot(no recognition on "Alexa" or "OK Google" (no custom wake word set). I renamed the file to original name and things worked on next boot.....I then instead renamed the GassistPi/src file to  GassistPi/backup.src and Amazon Alexa works but only with my newly created custom wake word "Oasis". Again, my question is about the GAssistPi folder (i ask because these are both your developments and  i see AssistantsPi did use the same "/home/pi/alexa-avs-sample-app" directory as alexa-avs-sample folder git would have. Is GassistPi part of the AssistantsPi setup or an unrelated folder?

shivasiddharth, 2018-01-08:
Please do not open two issues for the same query. Will answer your query in the other git.

closed 2018-01-08 by shivasiddharth

supreme83, 2018-01-09:
the question was related to this git initially and where it originates since it had to do with Gassist's placement once the other software is installed side by side.