shivasiddharth/GassistPi #123: Not sure what project and model IDs are and where I should put them

Ladyslayer0160, 2018-01-07:
At this step in the tutorial: "Open the service files in the /home/pi/GassistPi/systemd/ directory and add your project and model ids in the indicated places and save the file." I'm not sure where in the files I'm supposed to edit or what I'm supposed to write in them.

shivasiddharth, 2018-01-07:
In the line `ExecStart=/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id 'project-id'  --device_model_id 'saved-model-id'`

Replace 'project-id' and 'saved-model-id'. The project-id and model-id that you are supplying should also be within quotes.

Ladyslayer0160, 2018-01-08:
Alright, I found the project Id online on my Google account, but what is the saved-model-id, and where do I find it?

shivasiddharth, 2018-01-09:
your model-id will be in the home/pi directory in a text file

shivasiddharth, 2018-01-19:
Since i have not heard back, i am assuming that the issue has been resolved, so closing this issue for now, feel free to re-open this issue just in case.

closed 2018-01-19 by shivasiddharth