shivasiddharth/GassistPi #125: Raspberry PI Zero W Gassistant Problam

kumarmohan7, 2018-01-08:
Hi,

In My Raspberry Pi Zero W Gassistant Installation Problem,
In final Execution Stage this Error is Coming

Please give solution for solve problem....

INFO:root:Connecting to embeddedassistant.googleapis.com
WARNING:root:Device config not found: [Errno 2] No such file or directory: '/home/pi/.config/googlesamples-assistant/device_config.json'
INFO:root:Registering device
ERROR:root:Failed to register device: {
  "error": {
    "code": 400,
    "message": "Could not create the device instance. Client type need to be specified as SDK_LIBRARY or SDK_SERVICE.",
    "status": "INVALID_ARGUMENT"
  }
}

shivasiddharth, 2018-01-09:
If you already have a device registered and showing in the app, unlist/remove that device in the app and try registering the pi zero again.

kumarmohan7, 2018-01-11:
Hi,
Thank u for replay,
How to remove the device & more then one device enable any problem is coming  or any limitation is available...?

shivasiddharth, 2018-01-13:
I have already demonstrated the unlinking process in the youtube video :https://youtu.be/vprJBDDYE8Q

shivasiddharth, 2018-01-28:
Closing this issue !

closed 2018-01-28 by shivasiddharth