shivasiddharth/GassistPi #127: Update

coconuthero, 2018-01-09:
Hi Sid,

Sorry to bother you again but I am not familiar with github either...

I wanted to upgrade to the most recent version to enable google music streaming but I am not sure what to do with this script:
https://github.com/shivasiddharth/GassistPi/blob/update-script/GassistPi-09-Jan-2018-update.sh

Copy paste it into console?

shivasiddharth, 2018-01-09:
Download the script, uncomment the service that you are not using, by adding # symbol at the start of the line. Open a terminal and type: 
sudo chmod +x the path to your file
then 
sudo the path to your file

Take a look at the readme document, there are a number of commands like this.

coconuthero, 2018-01-09:
Hmm...so since there was no download option I copied the code and put it into a textfile that I named update and placed on the desktop. Trying to run it I get this:

pi@raspberrypi:~ $ sudo chmod +x /home/pi/Desktop/update
pi@raspberrypi:~ $ sudo /home/pi/Desktop/update
/home/pi/Desktop/update: 19: /home/pi/Desktop/update: Bad substitution

shivasiddharth, 2018-01-09:
There will be an option called RAW on right on the script page, clicking on that will open up the script in fullscreen mode on the browser, then right click and choose save as, just like how you save images from website.

coconuthero, 2018-01-09:
ok but now I get this:

pi@raspberrypi:~ $ sudo /home/pi/Desktop/GassistPi-09-Jan-2018-update.sh
This script must run as pi, trying to change user...
Failed to stop gassistpi-ok-google.service: Unit gassistpi-ok-google.service not loaded.

shivasiddharth, 2018-01-10:
Looks like you dont have the service enabled. So you dont have the google assistant running on boot ? I have already told you to comment the service if you are not running it, by adding # symbol before the start of the service lines.

coconuthero, 2018-01-10:
I get this error also when the assistant is running (manually enabled). But yes, I did not comment out anything since I don't understand the code. Looks like I've picked a mountain to big...but maybe a comment in the script which lines to comment out would be helpful to others like me as well...?

Btw I did amend my last 2 issue-posts (that you already closed) about language and headless start. Would be great if you could give your advice....

I am sorry to cause you all this issues...I understand now, that with my absence of knowledge about linux and raspberrys, I should not have started this project in the first place. But since I have already invested all this time, I would like to complete it and I am grateful for your advice and patience! 
Daniel

shivasiddharth, 2018-01-10:
No... Manually starting the assistant does not help. Please add a # symbol at the start of those two lines. It cannot be that tough. Or else, select those two lines, right click on the lines and choose the comment out option.

coconuthero, 2018-01-10:
Ok I finally got it all working! Thank you very much for your support!
One more question though: can I clone the SD card and use 2 or more Rasüberries simultaneusly?
And do you plan to integrate a "Play next song" feature for google music?

shivasiddharth, 2018-01-11:
Nope not advisable. You cannot have the same credentials/authentication for all your Pi

shivasiddharth, 2018-01-20:
Since the main issue has been resolved, closing this issue.

closed 2018-01-20 by shivasiddharth