shivasiddharth/GassistPi #128: Service restart when trying to stream music from YouTube?

Pancer1, 2018-01-10:
Hi! Does, anybody has a problem with "stream music from YouTube" functionality? Very often after I say "Stream <song and artist name>" I hear sound of starting GassistPi (start sound and Hi at the end) but nothing more, no music. It looks like GassistPi service is rebooting or something like that.

shivasiddharth, 2018-01-10:
Start the main.py manually and see as to what is causing the restart.

Pibreaker, 2018-01-15:
I've managed to get a stream from youtube but when I ask for a second the program give me the following error and closes:
"YouTube_No_Autoplay
    playshell.stdin.flush()
BrokenPipeError: [Errno 32] Broken pipe"

Might be related to the issue above...

shivasiddharth, 2018-01-19:
@Pibreaker , Is this issue only with the no autoplay function? Did you test the autoplay function ?

Pibreaker, 2018-01-20:
The issue occurred with both functions, yes. I mainly use the stream (no autoplay) function to stream long compilations so the autoplay function isn't used often. However, when it is used, the stop command stops the current music but moves on to the next one on the autoplaylist. It might be because of the modifications above...

shivasiddharth, 2018-01-20:
@Pibreaker , Yes, you need to kill mpsyt to stop streaming.  If you just stop vlc, then, mpsyt reads it as the end of file and start playing back the next one. So you really need to kill mpsyt.

I have never run into this issue of playing back after a stop command. Lets see as to how many of them report this.

alnug, 2018-01-21:
Hi Shivasiddharth,

First off huge thanks for starting this project. I too have experienced this issue. I did notice that i had not input my youtube API credentials as i was trying to debug this today. I just added and will test tomorrow to see if it still occurs.

closed 2018-01-28 by shivasiddharth

jabulonboaz, 2018-02-18:
Everything was working fie last week till this week i got the same issue, I say "Hey Google stream Frank Sinatra" or stream music or anything elese similar and then it says "getting YouTube link" and then I hear the GA initiation dingdong sound and nothing happens, here are my logs:

ON_MUTED_CHANGED:
  {'is_muted': False}
ON_START_FINISHED

ON_CONVERSATION_TURN_STARTED
ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {'text': 'stream Frank Sinatra'}
Getting youtube link
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
        version 1.23.8; written and copyright by Michael Hipp and others
        free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:01] Decoding of say.mp3 finished.
pi@raspberrypi:~ $

jabulonboaz, 2018-02-19:
Anyone please? Do we know why GA restarts when i ask it to stream ?
Should i update the git again?

PR #162 (2018-01-28): Cast YouTube videos to chromecast
1. Added logging of errors/crashes.
2. Changed text to speech conversion from Pico to GTTS combined with Google translate.
3. Improved streaming from YouTube.
4. Assigned mpv to play all media.
5. Modified installation script to install packages  by version number to prevent future breakdowns.
6. Added dipping of playing media to allow users to hear the Assistant responses better.
7. Added the media volume control and playback control.
8. Added YouTube casting to Chromecast feature.

Fixes  #160 , Fixes #158 , Fixes #161 , Fixes #156 , Fixes #150 , Fixes #148  [... cut]