shivasiddharth/GassistPi #129: Google assistant slow response

tstasevych, 2018-01-11:
Great project! Keep it going! 

One issue that i am facing however that after asking a question, it takes 10-15 seconds for response. Google Assistant on phone, or on Google Home works times faster. What is the reason behind it?

Thank you!

shivasiddharth, 2018-01-11:
It could be due to several reasons.
1. Lower processing power of Pi, when compared to other devices.
2. The multiple conditional statements that we have introduced for custom actions.
3. Network issues.

tstasevych, 2018-01-12:
How can I disable some of the conditional statements to test if that is the issue?

shivasiddharth, 2018-01-13:
Feel free to try whatever you feel like trying. Even if the script gets messed, anytime, you can get the script again from the git

shivasiddharth, 2018-01-28:
As i have not heard back, I am assuming that the issue has been resolved and closing this for now !

closed 2018-01-28 by shivasiddharth