shivasiddharth/GassistPi #131: Kodi with Google Assistant over local network

rkumarblr, 2018-01-13:
Hi Shivasiddharth,

My use case is to get google assistant to talk to kodi on my Nvidia Shield and do so within the local network and without use of IFTTT.
I am taking the liberty of writing to you since you have got it working on Pi

I have a google home mini and Nvidia shield with Kodi running on the shield.
Alternatively I can have kodi running on a Mac book pro and if I can control kodi on my Mac will also do.

So is the above possible ?
There is a user who has developed code to control kodi on the shield but that uses IFTTT and port forwarding which is not what I want to use due to issue with the ISP etc.

If its possible would it be  
Voice command to Google home mini -> Google Assistant (running on shield or Mac) ->  Node.js also running on shield or Mac ->  Kodi. Is that how it works ?
Your inputs is very much appreciated.

Thanks

shivasiddharth, 2018-01-13:
Without IFTTT, its not possible to control Kodi from Google Home Mini. If you want to locally control, then You need to run the Google Assistant on Pi and then control Kodi from Pi

rkumarblr, 2018-01-13:
Thanks. 
If you dont mind can you let me know in your above response, what exactly runs on the Pi that will enable me to control to kodi ? 
There are google assistant packages in  nodejs, so if its only the issue of assistant I should be able to do so on any PC that can run nodejs ? Is it the google action and where that runs that is the issue ?
Interestingly Shield has enabled Google Assistant and I can use the shield remote with a voice command to launch Kodi but nothing more I mean without the google mini but not aware if it can be configured to call custom google actions and how to interface between the two.

shivasiddharth, 2018-01-13:
I am using Google Assistant only for speech to text conversion. The controlling is done by Kodi Json-rpc api (python version). There is no reliable free text to speech conversion software for pcs. Linux x86 supports google assistant library. You can use either a standalone Linux pc or run a linux os in virtual box.

shivasiddharth, 2018-01-14:
Closing this issue !

closed 2018-01-14 by shivasiddharth

rkumarblr, 2018-01-16:
Shivasiddharth,

Could you please guide me on the high level flow of how the components
connect in your architecture and how they can work on a unix distribution
(preferably Mac)

My intent is to be able to control kodi through voice commands.Kodi running
on a mac book pro.
I currently achieved that as follows
GH mini -> IFTTT (Google Assistant and webhook) -> ngrok (a VPN since my
ISP blocked  external port) -> custom coded node.js (on the Mac book pro)
-> Kodi web control.
But this is by using IFTTT and only for Kodi.

I wish to control Kodi, Youtube and Google play music which you have
achieved.

So a high level flow like what I described above of how the various
components connect to each other and where they run will be helpful. The
details I can understand looking up the code once I understand the flow.

This is not an issue, since its technically not an issue with what you have
provide as a capability, its a different but related requirement.

Thanks

rkumarblr, 2018-01-16:
Just to further clarify my question.I do understand the limitation and only option for me if I want to use the GH Mini is to go through IFTTT. So if I do so what is the sequence of the flow of request to Kodi, YT and Google Play Music for controlling these three ser vices running on a linux distribution (or Mac)

shivasiddharth, 2018-01-16:
I am fetching tracks from GPM and playing them locally using gmusicapi. It does not amount to control. And similarly is the case for YT. I am converting YT videos to audio and playing them locally. I dont think that both of it can be done using IFTTT.

rkumarblr, 2018-01-17:
Thanks for the clarification