shivasiddharth/GassistPi #134: Setup on Pi3: Failed building wheel for grpcio / Missing wheel package

CelloVerb, 2018-01-16:
Hi there,

I've run into this a few times with the GassistPi setup for a Pi 3 - in the 
`python -m pip install --upgrade google-assistant-sdk[samples]` 
step, I get the following:
```
Requirement already up-to-date: setuptools in ./env/lib/python3.5/site-packages (from protobuf>=3.5.0.post1->grpcio>=1.3.5->google-assistant-grpc==0.1.0; extra == "samples"->google-assistant-sdk[samples])
Building wheels for collected packages: grpcio
  Running setup.py bdist_wheel for grpcio ... error
  Complete output from command /home/pi/env/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-u34ccx1k/grpcio/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmpra2n494_pip-wheel- --python-tag cp35:
  Found cython-generated files...
  usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
     or: -c cmd --help
  
  **error: invalid command 'bdist_wheel'**
  
  ----------------------------------------
  Failed building wheel for grpcio
  Running setup.py clean for grpcio
Failed to build grpcio
Installing collected packages: grpcio, googleapis-common-protos, google-assistant-grpc, pycparser, CFFI, sounddevice, monotonic, tenacity, futures
  Running setup.py install for grpcio ... \^canceled
Operation cancelled by user
```
And it gets stuck in the Running setup.py install for grpcio ... stage and has to be interrupted.

Doing a `pip install wheel` fixed it - might need that line added to the script.  Posting this here to save someone needing to figure that out.

This was on a fresh Raspbian Stretch image.

Cheers,
Cello

VinceW31, 2018-01-19:
During a new full install I also had a "Failed building wheel for .......(I cant remember now, but probably grpcio).  But the scrip continued anyway and completed OK.
If I now run "pip install wheel" as you suggest will that fix my installation, or will I then need to run all the scrips again afterwards?

shivasiddharth, 2018-01-19:
@VinceW31 You need not do anything, after throwing those errors, those components will indeed get installed. I have already talked about that in numerous places for more clearity check this out: https://youtu.be/vprJBDDYE8Q?t=709

@CelloVerb , Thanks for the shout. It will seem as though it gets stuck at grpcio, but its just that it will take time to install. I shall test the `pip install wheel` and included it in the next update. I am closing this issue for now to avoid confusion among newbies. Thanks.

closed 2018-01-19 by shivasiddharth

shivasiddharth, 2018-01-21:
@CelloVerb , #144 has has made the suggested changes, future users will not see the wheel build error.

prashantenterprisesblp, 2020-02-05:
I still see this error when installing on RPI.

SebbyBB, 2022-04-11:
Also still getting this issue on the Raspberry Pi, but to the point where (and not sure if this is directly because of this or an issue with the machine) the Pi crashes