shivasiddharth/GassistPi #136: How can I create a custom hotword for google assistant?

AshrafAkon, 2018-01-18:
Hey can you tell me how can i create a custom hotword for this assistant? I really need this and I am currently working on a project the needs to be finished. If someone knows how to change the default hotword then please let me know.

rottieGO, 2018-01-29:
Yes please!  Your previous version had snowboy included which worked great.  Why did you remove it in newest release?

moham96, 2018-01-31:
I'm waiting for this too.
@shivasiddharth  can you explain why did you remove snowboy and if you will add it in the future

shivasiddharth, 2018-01-31:
This new API runs into audio issues and crashes if not triggered properly, so had to remove it and moreover, i ported this project in less than 24hrs within google's update, had i been experimenting with custom wakeword, it would not have been possible.
I need to balance a lot of stuff for the custom wakeword, Pi Zero's CPU usage, triggering accuracy, audio issues etc, its not easy. Feature will be back, but i dont have an ETA for that.

moham96, 2018-02-01:
@shivasiddharth Thanks for clarifying the situation, In the meantime, do you think it's possible to get snowboy working using a workaround?according to the readme.md you used pin 22 to trigger the assistant, so my idea is to have Snowboy listening and when activated it would trigger another unused GPIO pin and by connecting this pin with pin 22 we can trigger the assistant. Do you think this can work? I don't have access to proper hardware to test it at the moment ( i will in a few weeks try to test it).keep up the good work.

nwaxiomatic, 2018-02-08:
any updates here?

SwannSchilling, 2018-02-10:
I would alsolove to have the Custom Wakeword feature back!! I am planning to implement the Google Assistant into a robot that I am building...so it would be very sad if I had to adress him with "OK Google"!!

I checked the news on updates and it seems like integrated custom wakewords are on their way, can you guys confirm this...should we just wait for the API to be updated?

The idea to have Snowboy triggering a GPIO Pin sounds very good too, but I guess @shivasiddharth took this into account already when updating his project?

moham96, 2018-02-10:
@SwannSchilling  Do you mean that this will be added by Google? I highly doubt it.Where did you read this?

SwannSchilling, 2018-02-10:
I am not sure if this is true, I just read two articles about this today...
I am actually highly interested in setting a custom hotword, I tried to get it done using snowboy...but I could not accomplish!! A tutorial on how to do this would be appreciated, I do not care too much about the streaming services @shivasiddharth was mentioning they would slow the system down, so using snowboy becomes unstable. I would just want to use my custom hotword to start the goole assistant!

Is that possible?

My sources were

https://chromeunboxed.com/news/custom-google-assistant-hotwords-soon/

https://9to5google.com/2018/01/29/google-app-7-20-assistant-apk-insight-teardown/

AshrafAkon, 2018-02-11:
Well I have successfully used google assistant sdk with a custom hotword and it is pretty stable. and you don't need to use gpio pins. If you know a little bit python then you just have to install snowboy from their official repo then use grpc to connect with the Api. If anyone wants the code i can help.

shivasiddharth, 2018-02-11:
@AshrafAkon , You can attach the file here/ paste the code. All those who want to experiment with custom wakeword can use that.

AshrafAkon, 2018-02-12:
https://bitbucket.org/dani_thomas/aiyhotworddetector/src 

these are the codes. I have tested this on aiy voice kit image, that uses python 3.4.2.

shivasiddharth, 2018-02-12:
I am aware of this piece of work by Dani for the AIY grp. There is a bit of difference between AIY and this project. Their scripts dont work here.

SwannSchilling, 2018-02-12:
I can use Snowboy to trigger the grpc demo of the voice assistan..and yes like @AshrafAkon said, it runs pretty stable!! But I do not know how to use this within gassistpi, anyony can help to set it up?
I would also need help on how to autostart Snowboy, and how to autostart gassistpi with button only an ok-google disabled! 

This is the code I am using to sart the original grpc demo of the AIY Kit...

SwannSchilling, 2018-02-12:
#!/usr/bin/env python3
# Copyright 2017 Google Inc.
#
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
#     http://www.apache.org/licenses/LICENSE-2.0
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.

"""A demo of the Google CloudSpeech recognizer."""

import aiy.audio
import aiy.cloudspeech
import aiy.voicehat

def main():
    recognizer = aiy.cloudspeech.get_recognizer()
    recognizer.expect_phrase('turn off the light')
    recognizer.expect_phrase('turn on the light')
    recognizer.expect_phrase('blink')

    button = aiy.voicehat.get_button()
    led = aiy.voicehat.get_led()
    aiy.audio.get_recorder().start()

    while True:
        print('Press the button and speak')
        button.wait_for_press()
        print('Listening...')
        text = recognizer.recognize()
        if not text:
            print('Sorry, I did not hear you.')
        else:
            print('You said "', text, '"')
            if 'turn on the light' in text:
                led.set_state(aiy.voicehat.LED.ON)
            elif 'turn off the light' in text:
                led.set_state(aiy.voicehat.LED.OFF)
            elif 'blink' in text:
                led.set_state(aiy.voicehat.LED.BLINK)
            elif 'goodbye' in text:
                break

if __name__ == '__main__':
    main()

AshrafAkon, 2018-02-12:
@SwannSchilling . I used the same algorithm to work on my project. You can use GassistPi's actions with this code and its pretty stable. Just a little bit of python. And the codes of the pushbutton.py of GassistPi and the code @SwannSchilling provided works the same way. In Dani's code you can add the actions of GassistPi.  You can also create your own actions. GassistPi is a great project. But the code performs little bit slow. I just use what i need in my code.

SwannSchilling, 2018-02-12:
@AshrafAkon Thanks, I will try to get this done!! Could you let me know how you autostart Snowboy? 
Because it needs to run this command...or is it set up like any othe autostart service?

sudo python trigger.py Kate.pmdl

First one ist my custom trigger.py placed it the snowboy directory, second it my custom hotword also within the snowboy directory!!

AshrafAkon, 2018-02-12:
@SwannSchilling are you talking about headless start? well you can use crontab or /etc/rc.local 
there are plenty of good tutorial of how to do that. But make sure to run this script inside a daemon(I actually don't know much about it) but you can use the script of the GassistPi for headless start. But make sure to change the path of the script. I am currently using the Dani's script. And I think with a little of editing I can use Dani's script to run on regular raspbian. Best of luck.

SwannSchilling, 2018-02-12:
I am having the AIY kit...I will check out how @shivasiddharth  and Dani made their startup scripts, I can surely get it done...thanks for the info, this is really a lot of fun!!! :)

aiemassfiria, 2018-03-13:
any progress news?

cucubits, 2018-03-20:
I would also love to see a guide of some sort to get this working. Ideally with the latest sdk. I cannot use it with the default wake word, it's just ...bad. Even alexa is a bit better on a rpi but it's not the best option. There's so many limitations there.

aiemassfiria, 2018-04-05:
@AshrafAkon Can i get the python script. I really need to use snowboy for my project.

aiemassfiria, 2018-04-11:
Successfully activate using snowboy with https://s3-us-west-2.amazonaws.com/snowboy/snowboy-releases/rpi-arm-raspbian-8.0-1.3.0.tar.bz2 (Precompile binary). Just added code to activate GPIO under **def play_audio_file(fname=DETECT_DING):** after **   audio.terminate()** in snowboydecoder.py . In Gassistpi/src/pushbutton.py i edited under **if wait_for_user_trigger:** at **if button_state==True:** to **if button_state==False:** . This is how my script works. Not pretty stable. But yes it works.

bancroftd, 2018-04-20:
hi all, can you post your script anyway. I would like to get it working with gassist pi or another image. I find with AIY images, kodi doesn't play all media, and I would like to get kodi working with voice commands.

Thanks

schromotion, 2018-04-22:
+1 for this feature! Been following this project for a while and it would be awesome to get this back

aiemassfiria, 2018-04-22:
This is the script that i modified.

http://199.175.49.14/snowboydecoder.py
http://199.175.49.14/pushbutton.py

In my case im using gpio12 not 22 so if u using gpio 22 change it back.

If someone have better code for editing pls share. Im not programmer...

rocketbhoom, 2018-06-18:
Do you _git_ it dude? I don't _git_ it cause' _git fetch && git checkout master_ ain't working for me. If you _git_ right it please tell me how to do it.

Thanks a lot,
Bhoom

PR #447 (2018-07-22): Custom wakeword activation for Armv7 devices
Fixes #136
closed 2018-07-22 by shivasiddharth