shivasiddharth/GassistPi #137: Assistant restarts when I try to stream from Google Music

iamericha, 2018-01-19:
For instance, I try the phrase:

"Hey Google, Play songs by artist RICHARD MARX from Google Music"

The assistant says it will try to find the songs from that artist but then restarts.

The youtube music play seems to work ok though.

Thanks!

shivasiddharth, 2018-01-19:
Have the google login credentials been provided in the gmusic.py script ? 
If yes, then stop the service using:
`sudo systemctl stop gassistpi-ok-google.service`
Manually run the main.py by executing the command given in the line "ExecStart=" in /home/pi/GassistPi/systemd/gassistpi-ok-google.service file, issue google music commands and post the error.

Please note that, the songs that you have in your phone does not amount to the songs in google play music library. You need to have songs https://play.google.com/music/listen?authuser&u=0#/recents

iamericha, 2018-01-19:
Thanks.  I may have misunderstood what I can do with this feature.  I won't be able to play songs from a radio station in my playlist in Google Music would I?

If I can, then I'll try your suggestion to get to the error.  I did enter my gmail id and password into the gmusic.py script too.  Can I run that command from the gassistpi-ok-google.service file via the terminal through SSH?  I don't have experience doing this so wanted to ask for some more guidance.

shivasiddharth, 2018-01-19:
Radio station playback is permitted only within US. I am outside US, so i dont have access to radio station playback feature. Normally i dont include stuff which i cannot test, because if someone comes back with some issue, there is no way i can help troubleshoot.

iamericha, 2018-01-19:
Thank you Sid. That would explain why it doesn’t work. I’ll stick to the YouTube playback for now.

lrsorey, 2018-01-21:
Hey Sid. First of all thanks for your contributions.

I too am having the same issue with Google Music as iamericha. Youtube streams fine and the Radio streams fine. I did execute your recommendation listed above and this is what I got:

$ python -u GassistPi/src/main.py --project_id 'My Project ID'  --device_model_id 'My Model ID'
  File "GassistPi/src/main.py", line 148
    print('Registering....', end='', flush=True)
                                            ^
SyntaxError: invalid syntax

I have my credentials in the gmusic.py api.login call.

Outside of Gmusic restarting everything works great.

shivasiddharth, 2018-01-21:
@lrsorey , You should execute it along with '/home/pi/env/bin/'

lrsorey, 2018-01-21:
Sorry about that Sid.

I did try  - > ExecStart=/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id 'Project Id'  --device_model_id 'Model ID' 

but I kept getting  - > -bash: -u: command not found

so I activated the virtual environment and ran the following:

python -u /home/pi/GassistPi/src/main.py --project_id 'Project ID Here'  --device_model_id 'Model ID Here'

a little strange, everything below is listed twice.

the following occurred:

(env) pi@raspberrypi:~ $ python -u /home/pi/GassistPi/src/main.py --project_id 'Project ID Here'  --device_model_id 'Model ID Here'
device_model_id: Device Model Listed Here
device_id: Device ID Listed Here

https://embeddedassistant.googleapis.com/v1alpha2/projects/ProjectIDListedHere/devices/DeviceIDListedHere 200
ON_MUTED_CHANGED:
  {'is_muted': False}
ON_MUTED_CHANGED:
  {'is_muted': False}
ON_START_FINISHED
ON_START_FINISHED

ON_CONVERSATION_TURN_STARTED
ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {'text': 'play songs by artist Richard Marx from Google Music'}
ON_RECOGNIZING_SPEECH_FINISHED:
  {'text': 'play songs by artist Richard Marx from Google Music'}
richard marx
ON_CONVERSATION_TURN_FINISHED:
  {'with_follow_on_turn': False}
ON_CONVERSATION_TURN_FINISHED:
  {'with_follow_on_turn': False} - (nothing plays after response from Google Assistant lookinhg for the artist richard marxx)

ON_CONVERSATION_TURN_STARTED
ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {'text': 'tell me about my day'}
ON_RECOGNIZING_SPEECH_FINISHED:
  {'text': 'tell me about my day'}
ON_RESPONDING_STARTED:
  {'is_error_response': False}
ON_RESPONDING_STARTED:
  {'is_error_response': False}
ON_RESPONDING_FINISHED
ON_RESPONDING_FINISHED
ON_CONVERSATION_TURN_FINISHED:
  {'with_follow_on_turn': False}
ON_CONVERSATION_TURN_FINISHED:
  {'with_follow_on_turn': False} - (Everything responds normal here)

ON_CONVERSATION_TURN_STARTED
ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {'text': 'stream Tears for Fears Everybody Wants to Rule the World'}
ON_RECOGNIZING_SPEECH_FINISHED:
  {'text': 'stream Tears for Fears Everybody Wants to Rule the World'}
Playing: tears for fears everybody wants to rule the world
ON_CONVERSATION_TURN_FINISHED:
  {'with_follow_on_turn': False}
ON_CONVERSATION_TURN_FINISHED:
  {'with_follow_on_turn': False} - (Youtube song plays here no problems)

ON_CONVERSATION_TURN_STARTED
ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {'text': 'stop'}
ON_RECOGNIZING_SPEECH_FINISHED:
  {'text': 'stop'}              - (Stop worked fine stopping the Youtube song)
ON_NO_RESPONSE
ON_NO_RESPONSE

shivasiddharth, 2018-01-21:
Do you have music uploaded to the google play music ? Do you see tracks by your artist here: https://play.google.com/music/listen?authuser&u=0#/artists ?
If you are trying to play radio Google Play music, please read my comment that i posted above, couple of days back.

lrsorey, 2018-01-21:
Ah, Ok.

I understand now.

theapu, 2018-02-03:
I am getting this error for "play songs by artist Daft Punk on google music" and assistant restarts. I have set api.login correctly in actions.py and I have artist Daftpunk at https://play.google.com/music/listen?authuser&u=0#/artists

Log file attached

[GassistPi.log](https://github.com/shivasiddharth/GassistPi/files/1692146/GassistPi.log)

Error:

```
2018-02-04 01:14:43,377 ERROR __main__ 'RequestPart' object is not callable
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 495, in <module>
    main()
  File "/home/pi/GassistPi/src/main.py", line 427, in main
    play_artist(artist)
  File "/home/pi/GassistPi/src/actions.py", line 1010, in play_artist
    tracks,numtracks=loadartist(artistname)
  File "/home/pi/GassistPi/src/actions.py", line 801, in loadartist
    songs_list= api.get_all_songs()
TypeError: 'RequestPart' object is not callable
```

shivasiddharth, 2018-02-03:
@theapu , Looks like the google account login credentials has not been entered.

theapu, 2018-02-04:
I have given correct login credentials in actions.py

`logged_in = api.login('<my gmail user name>@gmail.com', '<my app specific password>', Mobileclient.FROM_MAC_ADDRESS)`

When I run the below test.py (using code from actions.py) script after source ~/env/bin/activate I am able to generate songs.json and get stream url.  But when running GassistPi it crashes at gmusic api call api.get_all_songs() with above mentioned error and songs.json is not generated. I could not figure out whats the problem here.

```
from gmusicapi import Mobileclient
import json
import os
import os.path

api = Mobileclient()
api.login('<gmail user name>@gmail.com', '<my app sepcific password>', Mobileclient.FROM_MAC_ADDRESS)

song_ids=[]
track_ids=[]
albumname='random access memories'
def loadalbum(albumname):
    song_ids=[]
    album=str(albumname)
    if os.path.isfile("/home/pi/songs.json"):
        with open('/home/pi/songs.json','r') as input_file:
            songs_list= json.load(input_file)
##            print(songs_list)
    else:
        songs_list= api.get_all_songs()
        with open('/home/pi/songs.json', 'w') as output_file:
            json.dump(songs_list, output_file)
    for i in range(0,len(songs_list)):
        if album.lower() in (songs_list[i]['album']).lower():
            song_ids.append(songs_list[i]['id'])
        else:
            print("Album not found")
    songsnum=len(song_ids)
    return song_ids, songsnum

tracks,numtracks=loadalbum(albumname)

if os.path.isfile("/home/pi/.gmusicalbumplayer.json"):
	with open('/home/pi/.gmusicalbumplayer.json','r') as input_file:
		playerinfo= json.load(input_file)
		currenttrackid=playerinfo[0]

print(playerinfo)
print(tracks)
print(numtracks)
streamurl = api.get_stream_url(tracks[currenttrackid])
print(streamurl)
```

theapu, 2018-02-06:
I found the problem (I use aftership to track my packages)
I changed in actions.py
```
api = aftership.APIv4('<my aftership api>')
couriers = api.couriers.all.get()
```
to
```
asapi = aftership.APIv4('<my aftership api>')
couriers = asapi.couriers.all.get()
```
and
`
 text=api.trackings.get(tracking=dict(slug=slug, tracking_number=number))`
to
` text=asapi.trackings.get(tracking=dict(slug=slug, tracking_number=number))
`
and now google music works.

shivasiddharth, 2018-02-06:
@theapu , Thanks for pointing it out. Will fix the naming.

shivasiddharth, 2018-02-06:
As all the issues has been addressed, closing this thread.

closed 2018-02-06 by shivasiddharth

Akshaykkapoor, 2018-06-07:
ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play all songs from Google Music"}
stopping vlc
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}

 
Sidd , please help ....it says stopping vlc one the command to play google music is given ..