shivasiddharth/GassistPi #139: Repeat Installation issue on Raspi3  !!

fatubachan, 2018-01-19:
For the very first install went well  and even controlled Kodi  but after a while Gassistpi stopped responding but it responded to the wake word OK google or Hey google but there was no reply for the quires.

Post that I tried several times to re-install but ended with error like  "IsAdirectoryError: (Errorno 21) Is a directory: '/home/pi/' 

And had failed  building wheel for pyaudio ; failed building wheel for grpcio.

shivasiddharth, 2018-01-19:
The assistant will defenitely reply to query if it picked up the hotword. The volume of the updated assistant will be low on start. It will be too low and easy to miss i have already discussed that in #133 

Regarding the error, what you have mentioned does not convey much, only if you could paste the error as it is, i can try and help.

shivasiddharth, 2018-02-09:
Closing this due to inactivity

closed 2018-02-09 by shivasiddharth