shivasiddharth/GassistPi #140: "Hmm Something went wrong, try again in a few seconds"

abhishek96, 2018-01-20:
I setup Gassist on my RasPi3 and all seemed to work normally . However, one day later, the "Ok,Google" gets the Pi triggered but after that it either says "Hmm something went wrong , try again in a few seconds" or "There was a glitch". Can someone help? Thanks.

mashrurbd, 2018-01-21:
Not google server issue, it occurs when you call Assistant api more than 500 times a day, you'll be blocked from using google assistant for 24h. After this time period, you'll be okay to go with.

Zwillyzz, 2020-11-27:
Not google server issue, it occurs when you call Assistant api more than 500 times a day, you'll be blocked from using google assistant for 24h. After this time period, you'll be okay to go with.

Really?

Zwillyzz, 2020-11-27:
> Not google server issue, it occurs when you call Assistant api more than 500 times a day, you'll be blocked from using google assistant for 24h. After this time period, you'll be okay to go with.

Really?