shivasiddharth/GassistPi #143: Failed building wheel for grpcio

paolorotolo, 2018-01-21:
Log output:

```bash
Building wheels for collected packages: grpcio
Running setup.py bdist_wheel for grpcio ... error
Complete output from command /home/pi/env/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-1elj70wp/grpcio/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmp1ry607i3pip-wheel- --python-tag cp35:
Found cython-generated files...
usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
or: -c --help [cmd1 cmd2 ...]
or: -c --help-commands
or: -c cmd --help

error: invalid command 'bdist_wheel'

----------------------------------------
Failed building wheel for grpcio
Running setup.py clean for grpcio
Failed to build grpcio
```

closed 2018-01-21 by shivasiddharth

vinnyJager, 2018-02-05:
You could try installing wheel. `pip install wheel
` after I did that it worked, took some time though

RandallShanePhD, 2020-04-20:
This issue is back with Ubuntu 19.04 using python 3.7.3 and the pip installer.  Apologies for the length of the debug message here.  I cut out the part it did build else its war & peace!  Thx

`  ./src/core/lib/iomgr/ev_posix.h:46:16: warning: type ‘struct grpc_event_engine_vtable’ violates the C++ One Definition Rule [-Wodr]
   typedef struct grpc_event_engine_vtable {
                  ^
  ./src/core/lib/iomgr/ev_posix.h:46:16: note: a different type is defined in another translation unit
   typedef struct grpc_event_engine_vtable {
                  ^
  ./src/core/lib/iomgr/ev_posix.h:51:65: note: the first difference of corresponding definitions is field ‘fd_create’
     grpc_fd* (*fd_create)(int fd, const char* name, bool track_err);
                                                                   ^
  ./src/core/lib/iomgr/ev_posix.h:51:65: note: a field of same name but different type is defined in another translation unit
     grpc_fd* (*fd_create)(int fd, const char* name, bool track_err);
                                                                   ^
  src/core/lib/iomgr/ev_epoll1_linux.cc:141:8: warning: type ‘struct grpc_fd’ violates the C++ One Definition Rule [-Wodr]
   struct grpc_fd {
          ^
  src/core/lib/iomgr/ev_epollex_linux.cc:164:8: note: a different type is defined in another translation unit
   struct grpc_fd {
          ^
  src/core/lib/iomgr/ev_epoll1_linux.cc:144:58: note: the first difference of corresponding definitions is field ‘read_closure’
     grpc_core::ManualConstructor<grpc_core::LockfreeEvent> read_closure;
                                                            ^
  src/core/lib/iomgr/ev_epollex_linux.cc:237:19: note: a field with different name is defined in another translation unit
     gpr_atm refst = 1;
                     ^
  src/core/lib/iomgr/ev_epoll1_linux.cc:177:8: warning: type ‘struct grpc_pollset_worker’ violates the C++ One Definition Rule [-Wodr]
   struct grpc_pollset_worker {
          ^
  src/core/lib/iomgr/ev_epollex_linux.cc:273:8: note: a different type is defined in another translation unit
   struct grpc_pollset_worker {
          ^
  src/core/lib/iomgr/ev_epoll1_linux.cc:178:14: note: the first difference of corresponding definitions is field ‘state’
     kick_state state;
                ^
  src/core/lib/iomgr/ev_epollex_linux.cc:274:8: note: a field with different name is defined in another translation unit
     bool kicked;
          ^
  src/core/lib/iomgr/ev_epoll1_linux.cc:205:8: warning: type ‘struct grpc_pollset’ violates the C++ One Definition Rule [-Wodr]
   struct grpc_pollset {
          ^
  src/core/lib/iomgr/ev_epollex_linux.cc:287:8: note: a different type is defined in another translation unit
   struct grpc_pollset {
          ^
  src/core/lib/iomgr/ev_epoll1_linux.cc:207:25: note: the first difference of corresponding definitions is field ‘neighborhood’
     pollset_neighborhood* neighborhood;
                           ^
  src/core/lib/iomgr/ev_epollex_linux.cc:289:11: note: a field with different name is defined in another translation unit
     gpr_atm worker_count;
             ^
  src/core/lib/iomgr/ev_epoll1_linux.cc:230:8: warning: type ‘struct grpc_pollset_set’ violates the C++ One Definition Rule [-Wodr]
   struct grpc_pollset_set {
          ^
  src/core/lib/iomgr/ev_epollex_linux.cc:303:8: note: a different type is defined in another translation unit
   struct grpc_pollset_set {
          ^
  src/core/lib/iomgr/ev_epoll1_linux.cc:231:8: note: the first difference of corresponding definitions is field ‘unused’
     char unused;
          ^
  src/core/lib/iomgr/ev_epollex_linux.cc:304:23: note: a field with different name is defined in another translation unit
     grpc_core::RefCount refs;
                         ^
  src/core/lib/iomgr/ev_epoll1_linux.cc:488:45: warning: type ‘worker_remove_result’ violates the C++ One Definition Rule [-Wodr]
   typedef enum { EMPTIED, NEW_ROOT, REMOVED } worker_remove_result;
                                               ^
  src/core/lib/iomgr/ev_epollex_linux.cc:984:57: note: an enum with different value name is defined in another translation unit
   typedef enum { WRR_NEW_ROOT, WRR_EMPTIED, WRR_REMOVED } worker_remove_result;
                                                           ^
  src/core/lib/iomgr/ev_epoll1_linux.cc:135:8: warning: type ‘struct grpc_fork_fd_list’ violates the C++ One Definition Rule [-Wodr]
   struct grpc_fork_fd_list {
[... 157 lines cut ...]
  distutils.errors.DistutilsExecError: command 'x86_64-linux-gnu-g++' failed with exit status 1

  During handling of the above exception, another exception occurred:

  Traceback (most recent call last):
    File "/tmp/pip-install-ram15zel/grpcio/src/python/grpcio/commands.py", line 262, in build_extensions
      build_ext.build_ext.build_extensions(self)
    File "/usr/lib/python3.7/distutils/command/build_ext.py", line 449, in build_extensions
      self._build_extensions_serial()
    File "/usr/lib/python3.7/distutils/command/build_ext.py", line 474, in _build_extensions_serial
      self.build_extension(ext)
    File "/usr/lib/python3/dist-packages/setuptools/command/build_ext.py", line 199, in build_extension
      _build_ext.build_extension(self, ext)
    File "/usr/lib/python3.7/distutils/command/build_ext.py", line 559, in build_extension
      target_lang=language)
    File "/usr/lib/python3.7/distutils/ccompiler.py", line 717, in link_shared_object
      extra_preargs, extra_postargs, build_temp, target_lang)
    File "/usr/lib/python3.7/distutils/unixccompiler.py", line 217, in link
      raise LinkError(msg)
  distutils.errors.LinkError: command 'x86_64-linux-gnu-g++' failed with exit status 1

  ----------------------------------------
  Failed building wheel for grpcio
  Running setup.py clean for grpcio
Failed to build grpcio`

shivasiddharth, 2020-04-20:
@RandallShanePhD , For Ubuntu I have tested it only till the Bionic version. I have clearly mentioned that in the README. https://github.com/shivasiddharth/GassistPi#only-oses-suported-are

commit 44259922 (2018-01-21): Merge pull request #144 from paolorotolo/master

Add missing wheel dependency on installer. Closes #143
commit 8d4136da (2018-01-21): Add missing wheel dependency on installer. Closes #143