shivasiddharth/GassistPi #148: Streaming Radio problem after update 09/01/18

ifb777, 2018-01-22:
So the time has come again to seek advice from this list.
Having got most features working reasonably reliably over the past couple of weeks, I decided to update GassistPi to the 9th January version.
Since then, issuing the command "hey google,  Tune into, Radio One",  now just responds with...."Sorry, I don't know how to help with that".  The same result for successive stations.
Prior to the update, Streaming Radio using the default stations and some additional stations that I added worked well.
I have reviewed the code for this function in actions.py and can see no differences between the working version of the code and the updated version and as a non coder, I am now at a loss as to where to go from here.
Any thoughts or suggestions would be very welcome as this is a game changer for the project I'm working on.

My thanks in advance.

shivasiddharth, 2018-01-22:
Hope this #142 gives you some idea.

ifb777, 2018-01-22:
Yet again I must thank you for the prompt response to this issue.

I have had a play around with the words and found that "Tune in to" produces results albeit not as reliably as I would like.  I'm hoping a little more training in vocalising the command may improve the situation.  There is no doubt that "Tune into" worked well prior to the upgrade but I guess that's another story.
Thanks again for the input.  I'll leave this thread open for a while and play some more.

PR #162 (2018-01-28): Cast YouTube videos to chromecast
1. Added logging of errors/crashes.
2. Changed text to speech conversion from Pico to GTTS combined with Google translate.
3. Improved streaming from YouTube.
4. Assigned mpv to play all media.
5. Modified installation script to install packages  by version number to prevent future breakdowns.
6. Added dipping of playing media to allow users to hear the Assistant responses better.
7. Added the media volume control and playback control.
8. Added YouTube casting to Chromecast feature.

Fixes  #160 , Fixes #158 , Fixes #161 , Fixes #156 , Fixes #150 , Fixes #148  [... cut]
closed 2018-01-28 by shivasiddharth