shivasiddharth/GassistPi #149: [Question] Running next to Kodi

justinkruit, 2018-01-22:
Hello,

Amazing installation you got going here, way easier to set everything up correctly and with customization.

I'm running a Raspberry PI with OSMC (Kodi) on my room, and is it possible to actually install this on the same Raspberry, so both Kodi and GassistPi run simultaneously?

shivasiddharth, 2018-01-22:
You can do that, but i have drafted all the scripts for a pi user, so in all those places you need to change pi to osmc. I would suggest you to install the assistant on Raspbian Desktop and then install Kodi and set Kodi to auto start. There is no room for error if you take this route, IMHO.

shivasiddharth, 2018-01-23:
I have created a new branch for osmc users. You can try that and let me know as to how it goes.  [OSMC Branch](https://github.com/shivasiddharth/GassistPi/tree/osmc)

Pancer1, 2018-02-04:
@shivasiddharth Great Info! I'm using OSMC for years on another RP3. Now I can try to install GassistPi on it! OSMC branch will be updated with all new features in the same time as a main branch? If I remember you wrote that some features will not work with "no full desktop linux", because of lack of problems with some libraries. Is it actual?

shivasiddharth, 2018-02-04:
@Pancer1 , The standard Raspbian Lite does not come with certain packages, and also in Lite, there are issues with media streaming. Moreover, as you dont have a graphical environment in Lite, renaming, moving files etc need to be done using the terminal and first time users may find that difficult. That is why i ask ppl to use the Dekstop Image.
This OSMC branch is a test branch couple of commits behind the master. There has been no interest shown so far in installing the assistant alongside OSMC, so i have not bothered to update that. For sure there will be issues while installing the assistant in OSMC. If more and more people start using that and start reporting the errors, the branch can be developed faster. One person, going through cycles of erasing the cards and installing will take time. If ppl want the OSMC branch, i have no problems in updating it alongside the master.

Pancer1, 2018-02-04:
I will try to edit installer from latest build. Can You tell the features that are not working on Lite distributions? I think if those are Neopixel, nodeMCU or Parcel Tracking, I think some people (like me) don't use it.

shivasiddharth, 2018-02-04:
@BadFlo , Ok-google is pretty much working. Guess you are on the wrong thread. Maybe this is where you intended to post. #180

shivasiddharth, 2018-02-04:
@Pancer1 , Goolge Music, Music Volume Control, YouTube Streaming will not work as it is. By default, Lite does not come with python3, pip  so you may run into installation issues. Just make sure to add them in the installer that you are modifying. Also a number of features uses json, so also make sure to add that as well.

shivasiddharth, 2018-02-10:
@Pancer1 , I have tested the updated assistant and can confirm that all features work now in Lite version of Raspbian as well. Whatever additional packages were required, have been added to the requirements. Git package will not come pre-installed so just make sure to install that. Other than that, everything should be fine.

shivasiddharth, 2018-06-04:
Closing this for inactivity.

closed 2018-06-04 by shivasiddharth