shivasiddharth/GassistPi #152: only kind of an issue, only persue if it interests you

ErkFrr, 2018-01-23:
so, as i spoke with you about before, i have been trying to implement your code to call and run a specific CLI for pandora called Pianobar..   so, it works, but i havent figured out a way to control it yet.  the interesting thing ive run into is, with pianobar and gassistpi installed the google home assistants voice is so low and crackly its virtually inaudible..  now, i even started the whole project from scratch, same issue..  kinda odd, if you would like to take a crack at it, id appreciate your input.

thanks either way

shivasiddharth, 2018-01-23:
Frankly, i did not find time to explore Pianobar, but for the volume issue, the low volume of the assistant has already been discussed in #133

shivasiddharth, 2018-01-26:
Hello, Seems like Pandora is not available outside US, so unfortunately, there is nothing much i could do about the Pianobar.

shivasiddharth, 2018-02-09:
Closing this due to inactivity.

closed 2018-02-09 by shivasiddharth