shivasiddharth/GassistPi #154: portaudio19-dev install failed with "you have held broken packages."

cnhnlin, 2018-01-24:
Hi sir,

my os ver : raspbian stretch
Have you met this packages broken error messages!
How can i resolve those broken, any advice is helpful, thanks!!

my install steps is below:
$sudo apt-get install portaudio19-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 portaudio19-dev : Depends: libasound2-dev but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

 $ sudo apt-get install libasound2
Reading package lists... Done
Building dependency tree
Reading state information... Done
libasound2 is already the newest version (1.1.3-5+rpi3).
0 upgraded, 0 newly installed, 0 to remove and 44 not upgraded.

$ sudo apt-get install libasound2-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libasound2-dev : Depends: libasound2 (= 1.1.3-5)
E: Unable to correct problems, you have held broken packages.

BadFlo, 2018-01-24:
Hi,
Do you have the same error when you try from scratch?
I see you have 44 not updating packages, this would mean that you failed to
run sudo apt-get upgrade as asked earlier.

cnhnlin, 2018-01-25:
Thanks for the reply, I upgraded but it still broke errors there?! see below

$ sudo apt-get upgrade
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

$ sudo apt-get install portaudio19-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 portaudio19-dev : Depends: libasound2-dev but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

shivasiddharth, 2018-02-01:
@cnhnlin , 
Run `sudo apt-get update --fix-missing`

cnhnlin, 2018-02-01:
@shivasiddharth I'll try it! thanks

shivasiddharth, 2018-02-20:
Closing this for inactivity.

closed 2018-02-20 by shivasiddharth