shivasiddharth/GassistPi #155: No module named 'kodijson'

jagan7157, 2018-01-24:
hey sid,
            iam jagan ,i try to controll my rpi gpio through voice ..but i cant controll if iam try to say " trigger bedroom led on " its not working.
         and iam also running my main.py manuly its getting issue like
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 17, in <module>
    from kodijson import Kodi, PLAYER_VIDEO
ImportError: No module named 'kodijson'
        please help me sid....

shivasiddharth, 2018-01-24:
I guess you are running the main.py outside the environment. You need to run it within environment.
First execute the following and then run the main.py:
`source env/bin/activate`
And within environment also, when you run the main.py, it should be run with the project-id and model-id else it will again throw out errors.

Most likely, GA would not have deciphered your command correctly, that may be the reason the triggering is not working.

shivasiddharth, 2018-02-09:
As i have not heard back, i am assuming that this issue has been resolved and hence closing this.

closed 2018-02-09 by shivasiddharth

jbrepogmailcom, 2018-08-19:
Hello! Getting also error no module kodijson. Furthermore, when trying to execute source env/bin/activate I get error "-bash: env/bin/activate: No such file or directory"

I made clean install of latest github version on Raspbian Lite