shivasiddharth/GassistPi #156: Google Music play songs from artist problem

ifb777, 2018-01-25:
So I have now spent several hours working with the Play songs by artist.......from google music and have tested the vocalisation in both GA and the GPM app in my android phone.  Some artists play OK and others refuse to respond in GA whilst working in android.  I decided that I would run the troublesome artists by running main.py in the environment to see what is happening.
This is the outcome for one of the problem artists.  
`**Error in Voice Command hey google......play songs by artist John Lennon from google music**

Traceback (most recent call last):
  File "main.py", line 351, in <module>
    main()
  File "main.py", line 346, in main
    play_artist(artist)
  File "/home/pi/GassistPi/src/gmusic.py", line 247, in play_artist
    streamurl=api.get_stream_url(tracks[currenttrackid])
IndexError: list index out of range`

What does list index out of range mean?

I am now way out of my depth.  Any input would be most welcome

TIA

shivasiddharth, 2018-01-25:
The songs in google music as it is do not have playable links, they are represented by by a song-id. When you search for songs by an artist, it will yield an array of song-ids. The streaming urls are got one at a time for each of the corresponding song-ids. This is a limitation of the API, the streaming links are valid only for 90 secs and after which they expire, so we get the streaming link for the first song and at the end of the playback, we request the streaming link for the second song and this process goes on.

Say if we have an array of 8 song-ids and we are trying to get the url of 9th song-id, that is when the index out of range error occurs. To overcome that, i have already taken precautionary measures in the code:
```
if currenttrackid<numtracks:
        streamurl=api.get_stream_url(tracks[currenttrackid])
        streamurl=("'"+streamurl+"'")
        print(streamurl)
        os.system('mpv --really-quiet '+streamurl+' &')
    elif currenttrackid>numtracks and loopstatus=='on':
        currenttrackid=0
        nexttrackid=1
        loopstatus='on'
        playerinfo=[nexttrackid,loopstatus]
        with open('/home/pi/.gmusicartistplayer.json', 'w') as output_file:
            json.dump(playerinfo,output_file)
        streamurl=api.get_stream_url(tracks[currenttrackid])
        streamurl=("'"+streamurl+"'")
```
So if you are requesting the url of 9th song-id from an array of 8 song-ids, it automatically resets.
But in your case, we are getting the error when the program is trying to get the links for the first song-id, which would mean that the array must be empty or in other words there is no song matching the artist. 

To confirm my thoughts, you could try this:
open the gmusic.py script, scroll down to the `def play_artist(artistname):` function, and after the line `tracks,numtracks=loadartist(artistname)` add the following:
```
print(tracks)
print(numtracks)
```

The assistant on phone, checks for the songs by an artist which you have on phone, but the assistant on pi checks for songs in the cloud (play.google.com/music/listen?authuser&u=0#/mylibrary) so, if you dont  have the song in the cloud, that is when it returns an empty result.

ifb777, 2018-01-25:
Thanks for that information.
A thought just occurred to me that the the artist called in this issue was uploaded to GPM after I had been using GA for a few days. Could that have contributed to the indexing issue perhaps?
I have checked that all the problem music plays cleanly in the cloud by using Google Play Music Desktop Player on a PC that has no music stored locally.

Anyway, I will give your suggestion a try and post back.

Thanks again

ifb777, 2018-01-25:
I have followed your suggestion in your reply and have run main.py in the environment.  The following is a snapshot of the outcome:
`ON_RECOGNIZING_SPEECH_FINISHED:
  {'text': 'play songs by artist John Lennon from Google Music'}
ON_RECOGNIZING_SPEECH_FINISHED:
  {'text': 'play songs by artist John Lennon from Google Music'}
rm: cannot remove '/home/pi/GassistPi/src/trackchange.py': No such file or directory
rm: cannot remove '/home/pi/.gmusicartistplayer.json': No such file or directory
john lennon
Artist not found
Artist not found
Artist not found
Artist not found
Artist not found
Artist not found
Artist not found`

plus lots more of these.

Both trackchange.py and gmusicartistplayer.json exist in their respective locations.
Is this what you were  expecting to see?  Any further suggestions?
Thanks

shivasiddharth, 2018-01-25:
There will be two files in the /home/pi/ directory labled `songs.json` and `playlist.json`. Whenever you add or delete music to GPM cloud, make sure to delete those two files, that will refresh local references. I have infact added a function to refresh the list, but have not assigned a corresponding voice command, i shall do it in the next update.

ifb777, 2018-01-25:
Thanks for that.  I will get back to this little challenge tomorrow and post the outcome.

ifb777, 2018-01-27:
I post this update in the hope that it will serve to help others who are experiencing problems with playing music via the Google Play function.

In essence deleting both songs.json and playlist.json solved some of the problems but more importantly I discovered that those Albums that I uploaded to GPM that were ripped from CD did not have an "Artist Name associated with them.  This resulted in the Artist Name field in the Album record being blank.  So of course GassistPi's Google music code was never going to work because it was looking for something that didn't exist.  I really can be a "bunny" sometimes.

Anyway,  having fixed this using the "Edit Album" function in GPM to add Artist Name to the record all is now good and selecting either artist or album in the voice command works with 100% reliability so I am  a very happy chappy.

Thanks and accolades go to shivasiddharth for his work on the GassistPi project and for his input into the problem solving.  Now on to the next thing I dont understand...................>

PR #162 (2018-01-28): Cast YouTube videos to chromecast
1. Added logging of errors/crashes.
2. Changed text to speech conversion from Pico to GTTS combined with Google translate.
3. Improved streaming from YouTube.
4. Assigned mpv to play all media.
5. Modified installation script to install packages  by version number to prevent future breakdowns.
6. Added dipping of playing media to allow users to hear the Assistant responses better.
7. Added the media volume control and playback control.
8. Added YouTube casting to Chromecast feature.

Fixes  #160 , Fixes #158 , Fixes #161 , Fixes #156 , Fixes #150 , Fixes #148  [... cut]
closed 2018-01-28 by shivasiddharth