shivasiddharth/GassistPi #157: sudo  /home/pi/GassistPi/scripts/gassist-installer-pi3.sh

Sumeetgk, 2018-01-26:
there was some connection drop in my pi I reused this command sudo  /home/pi/GassistPi/scripts/gassist-installer-pi3.sh 

and then this errors comes out 
sudo  /home/pi/GassistPi/scripts/gassist-installer-pi3.sh 

now I cant install it again bcos of that error is there any other way to use it without formatting the memory card (bcos memory card consist of various other application too) 

Waiting for ur reply

shivasiddharth, 2018-01-26:
Open the installer script and delete the following lines and then try:
```
mkdir -p /home/pi/.config/mpv/scripts/
mv /home/pi/GassistPi/src/end.lua /home/pi/.config/mpv/scripts/end.lua
```

Sumeetgk, 2018-01-26:
thx it worked but the assistant is only waking up via voice command i want to trigger her by push button i manually connected the pushbutton to gpio 22 and one to ground nd when i press no response but when i say ok google it works...did u tried triggering via button bcos its not working

shivasiddharth, 2018-01-26:
You need to edit the systemd file for the pushbutton service like how you did for wakeword triggering and then start the corresponding service. All the details are available in the git README.
Since the issue has been resolved, closing this thread.

closed 2018-01-26 by shivasiddharth