shivasiddharth/GassistPi #159: pip3 not found

morrolinux, 2018-01-26:
Is it just me, or python3-pip should be added to installation scripts to make it work?

shivasiddharth, 2018-01-26:
Not required actually. What OS are you running ?

morrolinux, 2018-01-26:
I'm running the latest raspbian. What do you mean by "not required"? take a look at gassist-installer-pi-zero.sh row 45 for instance.
It will fail the script if pip3 is not installed.
Also, how can i "manually" test weather the assistant is workin or not, except for when the setup is finished or a service is supposed to make it start?

shivasiddharth, 2018-01-26:
I have written the codes for a desktop version of Raspbian, which comes pre-installed with pip, that is why i said that it is not required. Looks like you are using a Raspbian Lite version which does not have pip. I have clearly mentioned the same in README in bold, requesting people to use the desktop version, giving reasons for it.

Always you can check the status of the service by using the `sudo systemctl status` command.

morrolinux, 2018-01-26:
yup I've read the README and thought "Not all features and Custom wakeword" was ok for me with a Pi Zero W. Anyway I was asking how to invoke the assistant directly rather than relaying on the service. because I needed to read realtime the stdin/sterr and using a button to invoke it seems to not work for me ATM

shivasiddharth, 2018-01-26:
By default, we are supposed to press the "Enter" key to start a conversation. I have modified it to work with a pushbutton. gRPC API does not support assistant invoking by any other means. For more info you cna through this: https://developers.google.com/assistant/sdk/guides/library/python/

shivasiddharth, 2018-02-20:
Closing this for inactivity.

closed 2018-02-20 by shivasiddharth