shivasiddharth/GassistPi #163: After the update issues

Sumeetgk, 2018-01-28:
Youtube playback is not working i have shown the issue in this video
plz take a look at it if something went wrong correct me 

https://youtu.be/NRXVYSH7PO4

waiting for ur reply

shivasiddharth, 2018-01-28:
1. Is this issue with Voice activated assistant and pushbutton or only one?
2. Did you check the media volume?
3. Right click on the Pi directory, choose "Show Hidden" and open a file called .mediavolume.json and see the current volume.
4. Log files are deleted on every restart/shutdown, otherwise, you should have the Log File.

Sumeetgk, 2018-01-28:
1)This issue is occurring in both of the cases (pushbutton as well as voice) 
2)Yes the media volume is maximum but the service gets restarted so i dont think media volume is the issue 
3)There is no file in the default pi location called mediavolume.json
[image]

4) as i shown in the video i havent restarted the pie...i shown it just after the issue and there is no log 
   
waiting for ur reply

shivasiddharth, 2018-01-28:
Stop the service and manually run the main.py. You will definitely see a log then.  When it crashes check the log file and upload it here.

Sumeetgk, 2018-01-28:
Ok uploading the screenshot there is still no log file in tmp also available in screenshot 
[image]

shivasiddharth, 2018-01-28:
Nope, this not how main.py should be manually run. Open the systemd file that you modified, copy the line that says `ExecStart=` paste it in the terminal and execute.

Sumeetgk, 2018-01-28:
bro, I'm not that pro in all of this like u but I have managed to run the file u said 

pi@raspberrypi:~/GassistPi/src $ ExecStart=/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py  --project_id vernal-sunspot-160703  --device_model_id vernal-sunspot-160703-20180128085643
bash: -u: command not found

attaching screenshot for better reference 

[image]

shivasiddharth, 2018-01-28:
You should not include the ' ExecStart='. Just execute whatever is there after the = symbol.

Sumeetgk, 2018-01-28:
pi@raspberrypi:~ $ /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py  --project_id vernal-sunspot-160703  --device_model_id vernal-sunspot-160703-20180128085643

WARNING:root:pafy: youtube-dl not found; falling back to internal backend. This is not as well maintained as the youtube-dl backend. To hide this message, set the environmental variable PAFY_BACKEND to "internal".
INFO:gmusicapi.Mobileclient1:!-- begin debug log --!
INFO:gmusicapi.Mobileclient1:version: 11.0.0
INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
INFO:gmusicapi.Mobileclient1:initialized
INFO:gmusicapi.Mobileclient1:logged out
INFO:gmusicapi.Mobileclient1:authenticated
DEBUG:gmusicapi.Mobileclient1:Config(args=[], kwargs={})
DEBUG:gmusicapi.Mobileclient1:{'data': {'entries': [{'value': 'false', 'kind': 'sj#configEntry', 'key': 'canManageFP'}, {'value': 'true', 'kind': 'sj#configEntry', 'key': 'shouldShowLocalSuggestions'}, {'value': '3', 'kind': 'sj#configEntry', 'key': 'purchaseAccess'}, {'value': '30000', 'kind': 'sj#configEntry', 'key': 'wsVdtMillis'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'sideloadedPlaybackLogging'}, {'value': 'true', 'kind': 'sj#configEntry', 'key': 'isWarmWelcomeDisabledForWoodstock'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'canCancelSubscription'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'nashvillebg'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'isVsAvailable'}, {'value': 'true', 'kind': 'sj#configEntry', 'key': 'canBrowseSt'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'appstoreRatingPrompt'}, {'value': 'SUBSCRIBE', 'kind': 'sj#configEntry', 'key': 'upsellDrawerCtaMessage'}, {'value': "Download and listen to any song, album, or station when you're not connected", 'kind': 'sj#configEntry', 'key': 'upsellDrawerDownloadMessage'}, {'value': '5000', 'kind': 'sj#configEntry', 'key': 'wsVatMillis'}, {'value': 'true', 'kind': 'sj#configEntry', 'key': 'nuqRequiredForWsSignup'}, {'value': 'GET MORE', 'kind': 'sj#configEntry', 'key': 'upsellSkipLimitTwoLeftCtaMessage'}, {'value': 'SUBSCRIBE', 'kind': 'sj#configEntry', 'key': 'upsellNewReleasesCtaMessage'}, {'value': 'SUBSCRIBE', 'kind': 'sj#configEntry', 'key': 'upsellListenNowCtaMessage'}, {'value': 'true', 'kind': 'sj#configEntry', 'key': 'enableWsExtras'}, {'value': 'Subscribe to get millions of songs on-demand, offline, and ad-free', 'kind': 'sj#configEntry', 'key': 'upsellSignedOutMessage'}, {'value': '500', 'kind': 'sj#configEntry', 'key': 'topListMaxSize'}, {'value': '2 skips left', 'kind': 'sj#configEntry', 'key': 'upsellSkipLimitTwoLeftMessage'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'showLMark'}, {'value': 'true', 'kind': 'sj#configEntry', 'key': 'conciergeListenNow'}, {'value': '6', 'kind': 'sj#configEntry', 'key': 'minimumAppStartCountForFreeUpsell'}, {'value': 'SUBSCRIBE', 'kind': 'sj#configEntry', 'key': 'upsellPreviousCtaMessage'}, {'value': "Download and listen to any song, album, or station when you're not connected", 'kind': 'sj#configEntry', 'key': 'upsellDownloadMessage'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'showYoutubeRedInfoCard'}, {'value': 'true', 'kind': 'sj#configEntry', 'key': 'concListenNow'}, {'value': '300', 'kind': 'sj#configEntry', 'key': 'instantSearchDelay'}, {'value': 'Subscribe to play any song or album', 'kind': 'sj#configEntry', 'key': 'upsellChartsMessage'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'isPodcastSyncEnabled'}, {'value': 'SUBSCRIBE', 'kind': 'sj#configEntry', 'key': 'upsellChartsCtaMessage'}, {'value': 'true', 'kind': 'sj#configEntry', 'key': 'showViOSL'}, {'value': '604800', 'kind': 'sj#configEntry', 'key': 'topListUpdatePeriodSeconds'}, {'value': 'true', 'kind': 'sj#configEntry', 'key': 'isSuggestDeletionEnabled'}, {'value': 'true', 'kind': 'sj#configEntry', 'key': 'shouldShowServerSuggestions'}, {'value': '30', 'kind': 'sj#configEntry', 'key': 'situationsResponseTTLMinutes'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'isFoplessNautilusUser'}, {'value': '1459804296734', 'kind': 'sj#configEntry', 'key': 'appStartCountForFreeUpsellRevisionNumber'}, {'value': 'Choose your language preference here, too.', 'kind': 'sj#configEntry', 'key': 'mgRPBDesc'}, {'value': '1', 'kind': 'sj#configEntry', 'key': 'partridgeManagement'}, {'value': 'true', 'kind': 'sj#configEntry', 'key': 'showB'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'showA'}, {'value': 'true', 'kind': 'sj#configEntry', 'key': 'shouldEnableThriller'}, {'value': 'Subscribe to play any song or album', 'kind': 'sj#configEntry', 'key': 'upsellThumbsUpMessage'}, {'value': '100', 'kind': 'sj#configEntry', 'key': 'topListPageSize'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'isHallmarkEnabled'}, {'value': 'SUBSCRIBE', 'kind': 'sj#configEntry', 'key': 'upsellThumbsUpCtaMessage'}, {'value': '3', 'kind': 'sj#configEntry', 'key': 'uploadAccess'}, {'value': 'true', 'kind': 'sj#configEntry', 'key': 'isNautilusAvailable'}, {'value': '2', 'kind': 'sj#configEntry', 'key': 'nautilusAccess'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'isQuizEnabledForWoodstockOnSignup'}, {'value': 'true', 'kind': 'sj#configEntry', 'key': 'showO'}, {'value': '200000', 'kind': 'sj#configEntry', 'key': 'disabledConnectivityTimeoutMs'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'hasPControlSettings'}, {'value': 'false', 'kind': 'sj#configEntry', 'key': 'wasRecentlyInFoplessNautilus'}, {'value': 'true', 'kin [... cut ...]
    req = g.opener.open(url)
  File "/usr/lib/python3.5/urllib/request.py", line 451, in open
    req = Request(fullurl, data)
  File "/usr/lib/python3.5/urllib/request.py", line 269, in __init__
    self.full_url = url
  File "/usr/lib/python3.5/urllib/request.py", line 295, in full_url
    self._parse()
  File "/usr/lib/python3.5/urllib/request.py", line 324, in _parse
    raise ValueError("unknown url type: %r" % self.full_url)
ValueError: unknown url type: '/yts/jsbin/player-vflX4ueE4/en_US/base.js'

Sumeetgk, 2018-01-28:
when i run the command it worked and it started the assistant and when i say stream song name then the error occurs 

File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/init.py", line 36, in autodetect
from google.appengine.api import memcache
ImportError: No module named 'google.appengine'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/file_cache.py", line 33, in 
from oauth2client.contrib.locked_file import LockedFile
ImportError: No module named 'oauth2client.contrib.locked_file'

etc etc Like shown in above

shivasiddharth, 2018-01-28:
Seems like a pafy conflict.

Open a terminal and execute the following:
`sudo pip3 uninstall mps-youtube youtube-dl`

restart your pi and try again

Sumeetgk, 2018-01-28:
pi@raspberrypi:~ $ sudo pip3 uninstall mps-youtube youtube-dl
Cannot uninstall requirement mps-youtube, not installed

After restarting i tried again telling stream i got u and this errors comes up 
:- 

ON_RECOGNIZING_SPEECH_FINISHED:
  {'text': 'stream I got you'}
Getting youtube link
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.23.8; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:01] Decoding of say.mp3 finished.
WARNING:googleapiclient.discovery_cache:file_cache is unavailable when using oauth2client >= 4.0.0
Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/__init__.py", line 36, in autodetect
    from google.appengine.api import memcache
ImportError: No module named 'google.appengine'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/file_cache.py", line 33, in <module>
    from oauth2client.contrib.locked_file import LockedFile
ImportError: No module named 'oauth2client.contrib.locked_file'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/file_cache.py", line 37, in <module>
    from oauth2client.locked_file import LockedFile
ImportError: No module named 'oauth2client.locked_file'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/__init__.py", line 41, in autodetect
    from . import file_cache
  File "/home/pi/env/lib/python3.5/site-packages/googleapiclient/discovery_cache/file_cache.py", line 41, in <module>
    'file_cache is unavailable when using oauth2client >= 4.0.0')
ImportError: file_cache is unavailable when using oauth2client >= 4.0.0
['https://www.youtube.com/watch?v=uEJuoEs1UxY']
['https://www.youtube.com/watch?v=uEJuoEs1UxY']
ERROR:__main__:unknown url type: '/yts/jsbin/player-vflX4ueE4/en_US/base.js'
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 495, in <module>
    main()
  File "/home/pi/GassistPi/src/main.py", line 249, in main
    YouTube_No_Autoplay(str(usrcmd).lower())
  File "/home/pi/GassistPi/src/actions.py", line 1130, in YouTube_No_Autoplay
    youtubeplayer()
  File "/home/pi/GassistPi/src/actions.py", line 1072, in youtubeplayer
    audiostream,videostream=youtube_stream_link(tracks[currenttrackid])
  File "/home/pi/GassistPi/src/actions.py", line 310, in youtube_stream_link
    video = pafy.new(url)
  File "/home/pi/env/lib/python3.5/site-packages/pafy/pafy.py", line 122, in new
    return Pafy(url, basic, gdata, size, callback, ydl_opts)
  File "/home/pi/env/lib/python3.5/site-packages/pafy/backend_internal.py", line 42, in __init__
    super(InternPafy, self).__init__(*args, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/pafy/backend_shared.py", line 96, in __init__
    self._fetch_basic()
  File "/home/pi/env/lib/python3.5/site-packages/pafy/backend_internal.py", line 92, in _fetch_basic
    smaps, js_url, mainfunc = get_js_sm(watchinfo, self.callback)
  File "/home/pi/env/lib/python3.5/site-packages/pafy/backend_internal.py", line 430, in get_js_sm
    dbg_ref="javascript", file_prefix="js-")
  File "/home/pi/env/lib/python3.5/site-packages/pafy/backend_internal.py", line 367, in fetch_cached
    data = fetch_decode(url, "utf8")  # unicode
  File "/home/pi/env/lib/python3.5/site-packages/pafy/pafy.py", line 65, in fetch_decode
    req = g.opener.open(url)
  File "/usr/lib/python3.5/urllib/request.py", line 451, in open
    req = Request(fullurl, data)
  File "/usr/lib/python3.5/urllib/request.py", line 269, in __init__
    self.full_url = url
  File "/usr/lib/python3.5/urllib/request.py", line 295, in full_url
    self._parse()
  File "/usr/lib/python3.5/urllib/request.py", line 324, in _parse
    raise ValueError("unknown url type: %r" % self.full_url)
ValueError: unknown url type: '/yts/jsbin/player-vflX4ueE4/en_US/base.js'

shivasiddharth, 2018-01-28:
Here is the fix:
1. move into the environment using:
`source env/bin/activate`
2. Execute:
`pip install youtube-dl`

Now either restart the service or manually run the main.py. This will definitely solve your issue.

Sumeetgk, 2018-01-29:
Yes it worked Thankyou!

shivasiddharth, 2018-01-29:
closing this..

closed 2018-01-29 by shivasiddharth