shivasiddharth/GassistPi #164: dosent work after rebbot

daniel1584, 2018-01-28:
i folowed the instructions and after entering medel id and running auto start comands the assistant dosent work any more, need help

shivasiddharth, 2018-01-28:
Open the systemd file where you entered the ids and copy paste the line here which says `ExecStart`

daniel1584, 2018-01-28:
at the top or botton

daniel1584, 2018-01-28:
the services section looks like this
[Service]
Environment=VIRTUAL_ENV=/home/pi/env
Environment=PATH=/home/pi/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
ExecStart=/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id *****  --device_model_id assistant1-***-***

daniel1584, 2018-01-28:
i can add that the start sound just disipeard too.
is there a command tho see the status of the assistat 
or a comand to stop all services and restart them.

shivasiddharth, 2018-01-28:
The ids in the systemd file should be entered within single quotes.
Open a terminal and run the following command and paste the output:
`sudo systemctl status gassistpi-ok-google.service`

Also you can find a log called GassistPi log in the /tmp/ directory. Please upload the log file here.

Also i need some info on What model of Pi you are using  and what OS you have installed.

daniel1584, 2018-01-28:
pi@raspberrypi:~ $ sudo systemctl status gassistpi-ok-google.service
● gassistpi-ok-google.service - Google Assistant-Voice
   Loaded: loaded (/lib/systemd/system/gassistpi-ok-google.service; enabled; ven
   Active: active (running) since Sun 2018-01-28 13:06:29 UTC; 849ms ago
 Main PID: 1532 (python)
   CGroup: /system.slice/gassistpi-ok-google.service
           └─1532 /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --pr

Jan 28 13:06:29 raspberrypi systemd[1]: Started Google Assistant-Voice.
lines 1-8/8 (END)

daniel1584, 2018-01-28:
I have a pi3 running latest raspbian

daniel1584, 2018-01-28:
I can't find the log

shivasiddharth, 2018-01-28:
Please do not truncate the status message and remove the ids. It has no security risks.

daniel1584, 2018-01-28:
gassistpi-ok-google.service - Google Assistant-Voice
   Loaded: loaded (/lib/systemd/system/gassistpi-ok-google.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2018-01-28 13:17:17 UTC; 2s ago
 Main PID: 1301 (python)
   CGroup: /system.slice/gassistpi-ok-google.service
           └─1301 /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id assistant1-193511 --device_model_id assistant1-193511-20180128113723

Jan 28 13:17:17 raspberrypi systemd[1]: Started Google Assistant-Voice.
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
lines 1-8/8 (END)

shivasiddharth, 2018-01-28:
Like i said before, the ids should be within single quotes.
1. Disable the existing service using:
```
sudo systemctl stop gassistpi-ok-google.service
sudo systemctl disable gassistpi-ok-google.service
```
2. Edit the gassistpi-ok-google.service file in the GassistPi/systemd/ folder and make sure that the ids are placed within the single quotes
3. Go through this process again https://github.com/shivasiddharth/GassistPi#headless-autostart-on-boot-service-setup

daniel1584, 2018-01-28:
now it looks like this

gassistpi-ok-google.service - Google Assistant-Voice
   Loaded: loaded (/lib/systemd/system/gassistpi-ok-google.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Sun 2018-01-28 13:33:27 UTC; 50ms ago
  Process: 984 ExecStart=/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id assistant1-193511 --device_model_id assistant
 Main PID: 984 (code=exited, status=1/FAILURE)

Jan 28 13:33:27 raspberrypi systemd[1]: gassistpi-ok-google.service: Unit entered failed state.
Jan 28 13:33:27 raspberrypi systemd[1]: gassistpi-ok-google.service: Failed with result 'exit-code'.
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~

daniel1584, 2018-01-28:
[Unit]
Description=Google Assistant-Voice
Wants=network-online.target
After=network-online.target

[Service]
Environment=VIRTUAL_ENV=/home/pi/env
Environment=PATH=/home/pi/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
ExecStart=/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id "assistant1-193511"  --device_model_id "assistant1-193511-20180128113723"

WorkingDirectory=/home/pi/
StandardOutput=inherit
StandardError=inherit
Restart=always
User=pi

[Install]
WantedBy=multi-user.target

daniel1584, 2018-01-28:
but still it wont work
i have this error now
an 28 13:33:27 raspberrypi systemd[1]: gassistpi-ok-google.service: Unit entered failed state.
Jan 28 13:33:27 raspberrypi systemd[1]: gassistpi-ok-google.service: Failed with result 'exit-code'.

daniel1584, 2018-01-28:
like this
[Unit]
Description=Google Assistant-Voice
Wants=network-online.target
After=network-online.target

[Service]
Environment=VIRTUAL_ENV=/home/pi/env
Environment=PATH=/home/pi/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
ExecStart=/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id 'assistant1-193511'  --device_model_id 'assistant1-193511-20180128113723'

WorkingDirectory=/home/pi/
StandardOutput=inherit
StandardError=inherit
Restart=always
User=pi

[Install]
WantedBy=multi-user.target

daniel1584, 2018-01-28:
gassistpi-ok-google.service - Google Assistant-Voice
   Loaded: loaded (/lib/systemd/system/gassistpi-ok-google.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2018-01-28 13:51:05 UTC; 5s ago
 Main PID: 1143 (python)
   CGroup: /system.slice/gassistpi-ok-google.service
           └─1143 /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id assistant1-193511 --device_model_id assistant1-193511-20180128113723

Jan 28 13:51:10 raspberrypi python[1143]: INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
Jan 28 13:51:10 raspberrypi python[1143]: INFO:gmusicapi.Mobileclient1:initialized
Jan 28 13:51:10 raspberrypi python[1143]: INFO:gmusicapi.Mobileclient1:logged out
Jan 28 13:51:10 raspberrypi python[1143]: INFO:gmusicapi.Mobileclient1:failed to authenticate
Jan 28 13:51:10 raspberrypi python[1143]: Traceback (most recent call last):
Jan 28 13:51:10 raspberrypi python[1143]:   File "/home/pi/GassistPi/src/main.py", line 33, in <module>
Jan 28 13:51:10 raspberrypi python[1143]:     from actions import say
Jan 28 13:51:10 raspberrypi python[1143]:   File "/home/pi/GassistPi/src/actions.py", line 37, in <module>
Jan 28 13:51:10 raspberrypi python[1143]:     DEVELOPER_KEY = AIzaSyAudaN0vWSA03GPPf7KlYP4ak2-psArFo8
Jan 28 13:51:10 raspberrypi python[1143]: NameError: name 'AIzaSyAudaN0vWSA03GPPf7KlYP4ak2' is not defined
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
lines 1-17/17 (END)

shivasiddharth, 2018-01-28:
Again, there the YouTube API needs to be within single quotes, you have mentioned it without quotes. Your Google music may not be authenticating for two reason,
1. There also you may have given your credentials without quotes. or
2. You may be using two factor authentication. 
I have clearly given in the readme as to what you need to do if you are using two step-authentication. (https://github.com/shivasiddharth/GassistPi#music-streaming-from-google-music)

daniel1584, 2018-01-28:
now i have this error

{
  "error": {
    "code": 403,
    "message": "The request is missing a valid API key.",
    "status": "PERMISSION_DENIED"
  }
}

shivasiddharth, 2018-01-28:
Please upload the error log file here. It will be in the /tmp/ directory.

daniel1584, 2018-01-28:
i cant find the log

daniel1584, 2018-01-28:
i found thi googleplay music log 
!-- begin debug log --!
version: 11.0.0
logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
2018-01-28 13:11:20,950 - gmusicapi.Mobileclient1 (shared:74) [INFO]: initialized
2018-01-28 13:11:20,951 - gmusicapi.Mobileclient1 (shared:98) [INFO]: logged out
2018-01-28 13:11:22,218 - gmusicapi.Mobileclient1 (mobileclient:140) [INFO]: failed to authenticate

shivasiddharth, 2018-01-28:
The truncated  errors do not convey anything. Please post the entire output of the service status like how you did before.

daniel1584, 2018-01-28:
gassistpi-ok-google.service - Google Assistant-Voice
   Loaded: loaded (/lib/systemd/system/gassistpi-ok-google.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2018-01-28 16:45:13 UTC; 1min 44s ago
 Main PID: 1307 (python)
   CGroup: /system.slice/gassistpi-ok-google.service
           └─1307 /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id assistant1-193511 --device_model_id assistant1-193511-20180128113723

Jan 28 16:45:18 raspberrypi python[1307]: INFO:gmusicapi.Mobileclient1:logging to: /home/pi/.cache/gmusicapi/log/gmusicapi.log
Jan 28 16:45:18 raspberrypi python[1307]: INFO:gmusicapi.Mobileclient1:initialized
Jan 28 16:45:18 raspberrypi python[1307]: INFO:gmusicapi.Mobileclient1:logged out
Jan 28 16:45:18 raspberrypi python[1307]: INFO:gmusicapi.Mobileclient1:failed to authenticate
Jan 28 16:45:19 raspberrypi python[1307]: device_model_id: assistant1-193511-20180128113723
Jan 28 16:45:19 raspberrypi python[1307]: device_id: FB9B565CBBA4AFB4732B0DAD6812E22D
Jan 28 16:45:19 raspberrypi python[1307]: https://embeddedassistant.googleapis.com/v1alpha2/projects/assistant1-193511/devices/FB9B565CBBA4AFB4732B0DAD6812E22D 200
Jan 28 16:45:19 raspberrypi python[1307]: ON_MUTED_CHANGED:
Jan 28 16:45:19 raspberrypi python[1307]:   {'is_muted': False}
Jan 28 16:45:19 raspberrypi python[1307]: ON_START_FINISHED
~
~
~
~
~
~

shivasiddharth, 2018-01-28:
Your assistant is running perfectly fine. For gmusic api authentication, you need to check your credentials. Since it involves your email id and password, i cannot do much about it.
You just need to make sure that they are entered within single quotes and if you are using a two step authentication (Google Authentication app) to sign into your account, then you need to generate an app specific password.
There is nothing more that i or anyone else can really do/suggest. You have to take it up yourselves from here. Since the primary issue of assistant not working has been resolved, closing this issue.

closed 2018-01-28 by shivasiddharth

shivasiddharth, 2018-01-28:
Initially, after setup, the volume may be low so you may not hear the assistant, see point-6 here. https://github.com/shivasiddharth/GassistPi#continue-after-setting-up-audio

daniel1584, 2018-01-28:
it say hi on reboot

daniel1584, 2018-01-28:
i havent entered anything on play music but still have this issue

daniel1584, 2018-01-28:
Jan 28 17:56:22 raspberrypi python[999]: INFO:gmusicapi.Mobileclient1:logging to
Jan 28 17:56:22 raspberrypi python[999]: INFO:gmusicapi.Mobileclient1:initialize
Jan 28 17:56:22 raspberrypi python[999]: INFO:gmusicapi.Mobileclient1:logged out
Jan 28 17:56:23 raspberrypi python[999]: INFO:gmusicapi.Mobileclient1:failed to 
Jan 28 17:56:23 raspberrypi python[999]: device_model_id: assistant1-193511-2018
Jan 28 17:56:23 raspberrypi python[999]: device_id: FB9B565CBBA4AFB4732B0DAD6812
Jan 28 17:56:24 raspberrypi python[999]: https://embeddedassistant.googleapis.co
Jan 28 17:56:24 raspberrypi python[999]: ON_MUTED_CHANGED:
Jan 28 17:56:24 raspberrypi python[999]:   {'is_muted': False}
Jan 28 17:56:24 raspberrypi python[999]: ON_START_FINISHED

daniel1584, 2018-01-28:
this is the music sction on action file
#Google Music Declarations
song_ids=[]
track_ids=[]
api = Mobileclient()
#If you are using two-step authentication, use app specific password. For guidelines, go through README
logged_in = api.login('ENTER_YOUR_EMAIL_HERE', 'ENETER_YOUR_PASSWORD', Mobileclient.FROM_MAC_ADDRESS)

daniel1584, 2018-01-28:
soo why dose that error apear if i dont use it

daniel1584, 2018-01-28:
can i use the json file again or do i need too make a new one if i reinstal

shivasiddharth, 2018-01-28:
There is no error. I really dont understand as to what error you are talking about. Since you have not entered anything in the music section, it is not loggining into the API. So where is the error ?

daniel1584, 2018-01-28:
Jan 28 17:56:22 raspberrypi python[999]: INFO:gmusicapi.Mobileclient1:initialized
Jan 28 17:56:22 raspberrypi python[999]: INFO:gmusicapi.Mobileclient1:logged out
Jan 28 17:56:23 raspberrypi python[999]: INFO:gmusicapi.Mobileclient1:failed to authenticate
Jan 28 17:56:23 raspberrypi python[999]: device_model_id: assistant1-193511-20180128113723
Jan 28 17:56:23 raspberrypi python[999]: device_id: FB9B565CBBA4AFB4732B0DAD6812E22D
Jan 28 17:56:24 raspberrypi python[999]: https://embeddedassistant.googleapis.com/v1alpha2/projects/assistant1-193511/devices/FB9B565CBBA4AFB4732B0DAD6812E22D 200

daniel1584, 2018-01-28:
INFO:gmusicapi.Mobileclient1:failed to authenticate

shivasiddharth, 2018-01-28:
You have not entered your credentials in the music section. that is why it is not authenticating the API. This API has nothing to do with the Google Assistant. This API is required only for streaming music from Google Music.

daniel1584, 2018-01-28:
ok
i will try to reinstall.
thank you for taking the time. im new to this. 
keep the good work up.
sorry if i spell bad but i have reading and writing issues and english is not my first launguage.

shivasiddharth, 2018-01-28:
Nope, there is no need to re-install everything is working fine. Just enter your gmail id and password in the music section, the INFO:gmusicapi.Mobileclient1:failed to authenticate will go off

daniel1584, 2018-01-28:
i tryed that but the same error came up

daniel1584, 2018-01-28:
is it my gmail ****@gmail.com and the pasword to my google acount

daniel1584, 2018-01-28:
● gassistpi-ok-google.service - Google Assistant-Voice
   Loaded: loaded (/lib/systemd/system/gassistpi-ok-google.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2018-01-28 21:20:55 UTC; 1min 6s ago
  Process: 1018 ExecStart=/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id last-19
 Main PID: 1018 (code=exited, status=1/FAILURE)

Jan 28 21:20:55 raspberrypi systemd[1]: gassistpi-ok-google.service: Unit entered failed state.
Jan 28 21:20:55 raspberrypi systemd[1]: gassistpi-ok-google.service: Failed with result 'exit-code'.
Jan 28 21:20:55 raspberrypi systemd[1]: gassistpi-ok-google.service: Service hold-off time over, schedul
Jan 28 21:20:55 raspberrypi systemd[1]: Stopped Google Assistant-Voice.
Jan 28 21:20:55 raspberrypi systemd[1]: gassistpi-ok-google.service: Start request repeated too quickly.
Jan 28 21:20:55 raspberrypi systemd[1]: Failed to start Google Assistant-Voice.
Jan 28 21:20:55 raspberrypi systemd[1]: gassistpi-ok-google.service: Unit entered failed state.
Jan 28 21:20:55 raspberrypi systemd[1]: gassistpi-ok-google.service: Failed with result 'exit-code'.
~
~
~
~
~
~
~
~
~
~

daniel1584, 2018-01-28:
I will take a look tomorrow