shivasiddharth/GassistPi #166: problem with pushbutton.py

Enzoinventor, 2018-01-28:
when I press the button, the assistant listens to me only once and then stops.
Or when he says what I can do for you? it does not emit the sound that comes back to listening and freezes again.
I tried to manually execute the pushbutton.py file and in the console the sentence that I said appears and then it's enough.
Even premento ctrl + c does not exit the terminal.
I await your advice.
Thank you in advance

shivasiddharth, 2018-02-01:
Will look into it

shivasiddharth, 2018-02-02:
Can you replace the existing file with [this](https://github.com/shivasiddharth/GassistPi/blob/tasmota-wemo/src/pushbutton.py) and then check to see if the issue is fixed ?

Enzoinventor, 2018-02-02:
hi is the same file I own.
I noticed that as soon as I turn on the raspberry, I hear the mp3 file hi !! and when I press the button and say hello, he answers and then hangs everything even hey google no longer works. I also noticed that when you stop the CPU goes to 0 while it should be between 20/30 max as usual.

shivasiddharth, 2018-02-02:
Nope it's not the same file. It's from a Dev branch that I am currently working on.

Enzoinventor, 2018-02-02:
ok, I replaced the file but the problem persists as before

shivasiddharth, 2018-02-02:
ok, thanks

shivasiddharth, 2018-02-03:
Ok, I have fixed that issue. replace your existing file with [this](https://github.com/shivasiddharth/GassistPi/blob/tasmota-wemo/src/pushbutton.py).

Enzoinventor, 2018-02-03:
hi, i've replaced the file again but i get the same error.
At this point I reformat the sd and start from scratch in order to find the solution going for exclusion.
the file you edited is already inside Gassistpi, since I will have to recon the project?
thank you

shivasiddharth, 2018-02-03:
I tested the moded file and the error seemed to have been fixed. like i said before, its from the development branch, so even if you clone it again, you would not get that file. You will need to manually download and replace.

Enzoinventor, 2018-02-04:
hi, I've formatted and reinstalled everything but the problem persists.
can you show me how did you connect gpio22 as a trigger pin? because I noticed that the loop is still starting pushbutton.py.
thank you

shivasiddharth, 2018-02-04:
I have already mentioned this a couple of times. Re-installing does not work. The fixed or modified script is still in the test branch, i have not merged it. You need to manually download and replace.

Enzoinventor, 2018-02-04:
yes I know!!! I modified the file as you said before taking it from the branch

PR #190 (2018-02-09): Added control of Sonoff and NodeMCU devices running Sonoff-Tasmota
Fixes #147  
Fixes #165 
Fixes #166 
Fixes #174
closed 2018-02-09 by shivasiddharth