shivasiddharth/GassistPi #167: After update startup sound repeating continously

Shidiqadm, 2018-01-29:
I have seen your latest update and updated the gassist after updating, the startup sound continously repeating. I have tried to completly format my sd card and reinstalling google assistant, but having the same problem.
But the google assistant working fine, youtube not working, and trigger shutdown working fine 
Pls help with this issue

Shidiqadm, 2018-01-29:
https://youtu.be/e57oF_qmR1g
This is my error and i have uploaded it in youtube watch this and help me solving this  issue

shivasiddharth, 2018-01-29:
I guess you have enabled the pushbutton google assistant service as well without modifying the systemd file. That is why that service is getting restarted. To stop that, open terminal and execute:
```
sudo systemctl stop gassistpi-push-button.service
sudo systemctl disable gassistpi-push-button.service
```
For the youtube issue, please see this #163 (https://github.com/shivasiddharth/GassistPi/issues/163#issuecomment-361090246)

Shidiqadm, 2018-01-29:
Sorry thats not working

shivasiddharth, 2018-01-29:
What's not working ? Fix for YouTube or the restart issue ?
Post the status of the service using:

sudo systemctl status gassistpi-ok-google.service

Also attach the log. If you can manually run the assistant and then post the error like #163 I will have something to look and provide help. videos do not help.

Shidiqadm, 2018-01-30:
THE LOG FILES ARE

2018-01-29 17:46:19,894 DEBUG google.auth.transport.requests Making request: POST https://accounts.google.com/o/oauth2/token
2018-01-29 17:46:19,900 DEBUG requests.packages.urllib3.connectionpool Starting new HTTPS connection (1): accounts.google.com
2018-01-29 17:46:20,039 DEBUG google.auth.transport.requests Making request: POST https://accounts.google.com/o/oauth2/token
2018-01-29 17:46:20,046 DEBUG requests.packages.urllib3.connectionpool Starting new HTTPS connection (1): accounts.google.com
2018-01-29 17:46:20,959 DEBUG requests.packages.urllib3.connectionpool https://accounts.google.com:443 "POST /o/oauth2/token HTTP/1.1" 200 None
2018-01-29 17:46:21,007 INFO root Connecting to embeddedassistant.googleapis.com
2018-01-29 17:46:21,018 WARNING root Device config not found: [Errno 2] No such file or directory: '/home/pi/.config/googlesamples-assistant/device_config.json'
2018-01-29 17:46:21,019 INFO root Registering device
2018-01-29 17:46:21,020 DEBUG requests.packages.urllib3.util.retry Converted retries value: 3 -> Retry(total=3, connect=None, read=None, redirect=None, status=None)
2018-01-29 17:46:21,026 DEBUG requests.packages.urllib3.connectionpool Starting new HTTPS connection (1): embeddedassistant.googleapis.com
2018-01-29 17:46:21,180 DEBUG requests.packages.urllib3.connectionpool https://accounts.google.com:443 "POST /o/oauth2/token HTTP/1.1" 200 None
2018-01-29 17:46:21,208 DEBUG requests.packages.urllib3.util.retry Converted retries value: 3 -> Retry(total=3, connect=None, read=None, redirect=None, status=None)
2018-01-29 17:46:21,217 DEBUG requests.packages.urllib3.connectionpool Starting new HTTPS connection (1): embeddedassistant.googleapis.com
2018-01-29 17:46:22,190 DEBUG requests.packages.urllib3.connectionpool https://embeddedassistant.googleapis.com:443 "GET /v1alpha2/projects/sidpi-193506/devices/842D9E26959F51029FCC512B679C121E HTTP/1.1" 200 None
2018-01-29 17:46:22,277 DEBUG requests.packages.urllib3.connectionpool https://embeddedassistant.googleapis.com:443 "POST /v1alpha2/projects/project-id/devices HTTP/1.1" 403 None
2018-01-29 17:46:22,281 ERROR root Failed to register device: {
  "error": {
    "code": 403,
    "message": "Could not create the device instance. Account doesn't have permission to perform this operation for project project-id. See https://developers.google.com/assistant/sdk/guides/library/python/embed/register-device",
    "status": "PERMISSION_DENIED"
  }
}

2018-01-30 11:42:13,338 DEBUG google.auth.transport.requests Making request: POST https://accounts.google.com/o/oauth2/token
2018-01-30 11:42:13,345 DEBUG requests.packages.urllib3.connectionpool Starting new HTTPS connection (1): accounts.google.com
2018-01-30 11:42:13,684 DEBUG google.auth.transport.requests Making request: POST https://accounts.google.com/o/oauth2/token
2018-01-30 11:42:13,691 DEBUG requests.packages.urllib3.connectionpool Starting new HTTPS connection (1): accounts.google.com
2018-01-30 11:42:14,307 DEBUG requests.packages.urllib3.connectionpool https://accounts.google.com:443 "POST /o/oauth2/token HTTP/1.1" 200 None
2018-01-30 11:42:15,171 DEBUG requests.packages.urllib3.connectionpool https://accounts.google.com:443 "POST /o/oauth2/token HTTP/1.1" 200 None
2018-01-30 11:42:15,230 INFO root Connecting to embeddedassistant.googleapis.com
2018-01-30 11:42:15,241 WARNING root Device config not found: [Errno 2] No such file or directory: '/home/pi/.config/googlesamples-assistant/device_config.json'
2018-01-30 11:42:15,241 INFO root Registering device
2018-01-30 11:42:15,243 DEBUG requests.packages.urllib3.util.retry Converted retries value: 3 -> Retry(total=3, connect=None, read=None, redirect=None, status=None)
2018-01-30 11:42:15,250 DEBUG requests.packages.urllib3.connectionpool Starting new HTTPS connection (1): embeddedassistant.googleapis.com
2018-01-30 11:42:17,180 DEBUG requests.packages.urllib3.connectionpool https://embeddedassistant.googleapis.com:443 "POST /v1alpha2/projects/project-id/devices HTTP/1.1" 403 None
2018-01-30 11:42:17,184 ERROR root Failed to register device: {
  "error": {
    "code": 403,
    "message": "Could not create the device instance. Account doesn't have permission to perform this operation for project project-id. See https://developers.google.com/assistant/sdk/guides/library/python/embed/register-device",
    "status": "PERMISSION_DENIED"
  }
}

2018-01-30 11:42:26,593 DEBUG google.auth.transport.requests Making request: POST https://accounts.google.com/o/oauth2/token
2018-01-30 11:42:26,599 DEBUG requests.packages.urllib3.connectionpool Starting new HTTPS connection (1): accounts.google.com
2018-01-30 11:42:29,554 DEBUG requests.packages.urllib3.connectionpool https://accounts.google.com:443 "POST /o/oauth2/token HTTP/1.1" 200 None
2018-01-30 11:42:29,581 INFO root Connecting to embeddedassistant.googleapis.com
2018-01-30 11:42:29,592 WARNING root Device config not found: [Errno 2] No such file or directory: '/home/pi/.config/googlesamples-assistant/device_config.json'
2018-01-30 11:42:29,592 INFO root Registering device
2018-01-30 11:42:29,594 DEBUG requests.packages.urllib3.util.retry Converted retries value: 3 -> Retry(total=3, connect=None, read=None, redirect=None, status=None)
2018-01-30 11:42:29,600 DEBUG requests.packages.urllib3.connectionpool Starting new HTTPS connection (1): embeddedassistant.googleapis.com
[... cut ...]
2018-01-30 11:52:26,422 DEBUG requests.packages.urllib3.connectionpool https://embeddedassistant.googleapis.com:443 "POST /v1alpha2/projects/project-id/devices HTTP/1.1" 403 None
2018-01-30 11:52:26,429 ERROR root Failed to register device: {
  "error": {
    "code": 403,
    "message": "Could not create the device instance. Account doesn't have permission to perform this operation for project project-id. See https://developers.google.com/assistant/sdk/guides/library/python/embed/register-device",
    "status": "PERMISSION_DENIED"
  }
}

Shidiqadm, 2018-01-30:
YouTube also not working even after 
Source env/bin/activate
Pip install youtube-dl

shivasiddharth, 2018-01-30:
Your device has not been registered. Re-run the installer script or follow the steps given here to register the device. https://developers.google.com/assistant/sdk/guides/library/python/embed/register-device

Shidiqadm, 2018-01-31:
Restart issue has been solved and what to do for YouTube streaming,it was not working as you said in #163

Shidiqadm, 2018-01-31:
(env) pi@raspberrypi:~ $ /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id sidpi-193506  --device_model_id sidpi-193506-20180131115457
device_model_id: sidpi-193506-20180131115457
device_id: 842D9E26959F51029FCC512B679C121E

https://embeddedassistant.googleapis.com/v1alpha2/projects/sidpi-193506/devices/842D9E26959F51029FCC512B679C121E 200
ON_MUTED_CHANGED:
  {'is_muted': False}
ON_START_FINISHED

ON_CONVERSATION_TURN_STARTED
ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {'text': 'stream I got you'}
Getting youtube link
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.23.8; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:01] Decoding of say.mp3 finished.

shivasiddharth, 2018-01-31:
Attach the log please. You have an option to attach files by selecting them, so attach the log and dont paste it.

Shidiqadm, 2018-01-31:
Sorry bro for distrub you ,the problem is I have not included my YouTube API in actions.py so I have fixed it 
Any way thankyou bro

Shidiqadm, 2018-01-31:
Can I change the wake word for Google assistant

And please tell me how in a clear way

BadFlo, 2018-01-31:
No, you cannot for the moment, this was possible with older API, but he is
having issues making it work for new API.

Shidiqadm, 2018-01-31:
Hmmm thank you for your information

shivasiddharth, 2018-01-31:
@BadFlo , Thanks.

Issue resolved so closing this.

closed 2018-01-31 by shivasiddharth