shivasiddharth/GassistPi #169: Youtube streaming problem

adwinang, 2018-01-29:
At the start, stream youtube had no problem, but I don't know what I did, but afterwards, the function just cannot run anymore. I get this problem 
`socat[1824] E connect(7, AF=1 "/tmp/mpvsocket", 16): Connection refused`
I rerun the program, i get the same problem again.
I reboot raspberry pi, but it says it's getting the youtube audio, but it takes forever, after waiting for quite awhile, it still doesn't get/run the audio.

shivasiddharth, 2018-01-29:
Right click on the Pi directory, choose "Show Hidden" and open a file called .mediavolume.json and see the number. if its 10, change it to 90, close and save.
After that try giving the streaming command.

adwinang, 2018-01-29:
Ah, it worked now, Thank you!

shivasiddharth, 2018-01-29:
Ok, let me give you the reason, so that others can also understand as to why this happened.
When the assistant is triggered, i have set the volume to be pulled down to 10 for obvious reasons. If the conversation terminated the normal way, then the assistant will read the volume from the file and set it up back again. 
But if the assistant crashes for some reason in-between the media volume stays at 10 and during subsequent triggering, the 10 will be written on to the file as the existing normal volume.

closed 2018-01-29 by shivasiddharth