shivasiddharth/GassistPi #170: trackchange.py not found

mwhitacre, 2018-01-30:
during playing of radio stream, pushbutton.py calls for trackchange.py but file cannot be found

shivasiddharth, 2018-01-31:
radio stream does not have an end of file event so the trackchange.py file is not required. Since the mpv will run in background, it will not crash the assistant. So you can ignore that.

mwhitacre, 2018-02-01:
this is my error

ON_CONVERSATION_TURN_STARTED
ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {'text': 'tune in to radio 3'}
50.000000
http://cristina.torontocast.com:8111
Tuning into Radio 3
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.23.8; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:02] Decoding of say.mp3 finished.
ON_CONVERSATION_TURN_FINISHED:
  {'with_follow_on_turn': False}
50.000000
2018/02/01 02:24:33 socat[13586] E connect(6, AF=1 "/tmp/mpvsocket", 16): Connection refused

/home/pi/env/bin/python: can't open file '/home/pi/GassistPi/src/trackchange.py': [Errno 2] No such file or directory

shivasiddharth, 2018-02-01:
The url for the stream is in the wrong format. The correct url is `http://cristina.torontocast.com:8111/listen.pls` For the shoutcast streams, add ` /listen.pls`

closed 2018-02-01 by shivasiddharth