shivasiddharth/GassistPi #174: Google Music Playlist behavior

ifb777, 2018-02-01:
Have just encountered this today. When google music reaches the end of the last track in a playlist,  any further attempts to issue the "Play songs from first playlist in Google Music" command is met with silence until the file /home/pi/playlist.json is deleted and the Pi is rebooted.  Normal Playlist activity the ensues.

Comments and or advice welcome

shivasiddharth, 2018-02-01:
Congrats you have earned a badge for testing and making this build better 🥇. JK! Open the actions.py file and scroll to the function, `def play_playlist(playlistnum): `  And within the `if not tracks==[]:` loop there will be a line `elif currenttrackid>numtracks and loopstatus=='on':` change that to 
`elif currenttrackid>numtracks:` and see if it solves your issue. Will roll out a fix over the weekend.

shivasiddharth, 2018-02-02:
Can you attach the log, that will provide more insight as to whats going on when all the tracks are done playing.

ifb777, 2018-02-04:
Sorry for the delay in responding. Here's what I have done.
I created a new Playlist 1 with three songs in it.
I deleted playlist.json and rebooted
This is the log with my comments between### ###
`###Process Starts###

ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {'text': 'play songs from First playlist in Google Music'}
ON_RECOGNIZING_SPEECH_FINISHED:
  {'text': 'play songs from First playlist in Google Music'}
'https://r1---sn-u2bpouxgoxu-ntqk.c.doc-0-0-sj.sj.googleusercontent.com/videoplayback?id=28906106697b9c44&itag=25&source=skyjam&begin=0&ei=RsJ2WpChD8aM2QSE1bWYBA&o=10808441032733454745&ratebypass=yes&cpn=RmGXMpayZ4fclRh_tttgcg&ip=14.200.121.149&ipbits=0&expire=1517732512&sparams=ei,expire,id,initcwndbps,ip,ipbits,itag,mm,mn,ms,mv,o,pcm2cms,pl,ratebypass,source&signature=124A44C75C946E3D0693484D386293C71A04F647.03BDEC9D613BD416F77B4CE612A873BD8B25832C&key=cms1&initcwndbps=5140&mm=31&mn=sn-u2bpouxgoxu-ntqk&ms=au&mt=1517732335&mv=m&pcm2cms=yes&pl=24'
ON_CONVERSATION_TURN_FINISHED:
  {'with_follow_on_turn': False}
ON_CONVERSATION_TURN_FINISHED:
  {'with_follow_on_turn': False}

###First song Ends (actually second song in playlist)###`

I didn't realise until I did this that the first song in the playlist doesn't play at all and the process starts with the second song.

###Second song Starts###

'https://sj.googleusercontent.com/stream?id=a5a55a988d75c678&itag=25&source=skyjam_user_upload&begin=0&o=10808441032733454745&ip=0.0.0.0&ipbits=0&expire=1517732678&sparams=id,itag,source,o,ip,ipbits,expire&signature=947AE3E91A377E9FF0D6677F62D8978D438960F9.C191F5F770A56BA2445A9FBA19F078125BFEC834&key=sj3'

###No further activity or Music. process hangs at this point###``

I now press the STOP Button and issue the voice command again (without touching Playlist.json)
This is the result
`### Attempt to play from playlist again ###

ON_START_FINISHED

ON_CONVERSATION_TURN_STARTED
ON_CONVERSATION_TURN_STARTED
ON_END_OF_UTTERANCE
ON_END_OF_UTTERANCE
ON_RECOGNIZING_SPEECH_FINISHED:
  {'text': 'play songs from the first playlist in Google Music'}
ON_RECOGNIZING_SPEECH_FINISHED:
  {'text': 'play songs from the first playlist in Google Music'}
'https://r7---sn-u2bpouxgoxu-ntqk.c.doc-0-0-sj.sj.googleusercontent.com/videoplayback?id=2e02088dda767486&itag=25&source=skyjam&begin=0&ei=OMZ2WsjeLMSK2QSK6IKADQ&o=10808441032733454745&ratebypass=yes&cpn=O3mG_LjoQKnUetYDL1C-fg&ip=14.200.121.149&ipbits=0&expire=1517733522&sparams=ei,expire,id,initcwndbps,ip,ipbits,itag,mm,mn,ms,mv,o,pcm2cms,pl,ratebypass,source&signature=30C5097606D7C6B4D1F10C7450D26FC06045934D.2CB57582F87CAF0AD1D44031D4E7BC2189640101&key=cms1&initcwndbps=5050&mm=31&mn=sn-u2bpouxgoxu-ntqk&ms=au&mt=1517733343&mv=m&pcm2cms=yes&pl=24'
sh: 1: mpv --really-quiet +streamurl+ &: not found
ON_CONVERSATION_TURN_FINISHED:
  {'with_follow_on_turn': False}
ON_CONVERSATION_TURN_FINISHED:
  {'with_follow_on_turn': False}

###The process ends here no Music Plays until playlist.json is deleted and the pi is Rebooted###`

I hope that this helpful.  Please let me know if I can do anything else to get the Google Music function working as I am sure you expect.

Thanks

shivasiddharth, 2018-02-04:
The starting of songs midway in playlist is intentional. Something like playing where you left. If you are playing song 1 and stopping the playback, it will start from song 2. Can you try the previous suggestion i gave and see if the tracks are looping back?

ifb777, 2018-02-05:
Ok... after I realised that I was not using the very latest version of GassisPi and corrected that, I made the suggested change to the actions.py file and removed the phrase "loopstatus=="on':  Hey presto! Google Music Playlist behaves well.  It starts playing with track 1 and when it reaches the end of the last track issuing "Play Songs from the first playlist in Google Music" starts the playlist from the beginning again which I assume is the expected behavior.

**On another matter,** with regard to calling multiple playlists. The "ReadMe" alludes to needing to modify the code to utilise 2nd, 3rd, 4th etc playlist.  Could you please expand on that for a non coder.  This would improve my project significantly.

Thanks in advance

shivasiddharth, 2018-02-05:
Ok, i shall fix the auto-looping in the next update and also will add a second playlist call. Let me know if you come across anything else which needs refinement or fix.

ifb777, 2018-02-05:
OK Thanks

**Could you look at this and provide some guidance**

**On another matter**, with regard to calling multiple playlists. The "ReadMe" alludes to needing to modify the code to utilise 2nd, 3rd, 4th etc playlist. Could you please expand on that for a non coder. This would improve my project significantly.

Thanks in advance

shivasiddharth, 2018-02-05:
Copy lines 374-377 in main.py and paste it before the `else:` make sure to allign the `if` Change instances of `play_playlist(0)` to `play_playlist(1)`, change 'first' to 'second', similarly 'one' and '1' in the newly pasted lines. 
Do the same in line 436 as well.

ifb777, 2018-02-05:
Thanks for that.  I'll give it a go tomorrow

VinceW31, 2018-02-06:
@shivasiddharth, Fixed
 within the same "if not tracks==[]:" loop, 
change 
"elif currenttrackid>numtracks and loopstatus=='on':" 
to
 "elif currenttrackid>=numtracks and loopstatus=='on':"
it works fine then

shivasiddharth, 2018-02-07:
@VinceW31 , Thanks, will add it to the next update.

ifb777, 2018-02-09:
Thanks to **VinceW31** the small code change posted a couple of days ago improved playlist behavior again ensuring that playlists cycle.  
I have now got multiple playlists working well with one small item to be fixed.
Thanks again

PR #190 (2018-02-09): Added control of Sonoff and NodeMCU devices running Sonoff-Tasmota
Fixes #147  
Fixes #165 
Fixes #166 
Fixes #174
closed 2018-02-09 by shivasiddharth