shivasiddharth/GassistPi #175: How to turnoff the autostart?

nitrogen73, 2018-02-01:
Im a beginner and i would like to know how to turn off the autostart at startup and how to start it manually.
Also i would like to know the status like in the terminal like as in MMM-googleassistant.
Anyone please help me?

VinceW31, 2018-02-04:
To Stop the service press Ctrl+C to get the command prompt, then enter the following command:
sudo systemctl stop gassistpi-ok-google.service

To restart again manually enter the same command you already have at the line starting with "ExecStart=" in /home/pi/GassistPi/systemd/gassistpi-ok-google.service file.

shivasiddharth, 2018-02-05:
stop command only temporarily stops the service. To completely disable the auto start,
`sudo systemctl disable gassistpi-ok-google.service`

closed 2018-02-05 by shivasiddharth