shivasiddharth/GassistPi #176: argument --device_model_id is required

jagan7157, 2018-02-01:
hey sid...
       sorry for disturbance ...iam getting this type of error how to i solve ....

usage: main.py [-h] [--credentials OAUTH2_CREDENTIALS_FILE] --device_model_id
               DEVICE_MODEL_ID [--project_id PROJECT_ID]
main.py: error: argument --device_model_id is required

BadFlo, 2018-02-01:
Hello,

Did you replace project ID and model device ID in the files as required by
the procedure?

jagan7157, 2018-02-01:
yes .. see this

Description=Google Assistant-Voice
Wants=network-online.target
After=network-online.target

[Service]
Environment=VIRTUAL_ENV=/home/pi/env
Environment=PATH=/home/pi/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
ExecStart=/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id device-192519  --device_model_id device-192519-20180131171828

WorkingDirectory=/home/pi/
StandardOutput=inherit
StandardError=inherit
Restart=always
User=pi

[Install]
WantedBy=multi-user.target

shivasiddharth, 2018-02-01:
You should add that within single quotes. Like how i have given in the default systemd file. And it should be a double space after the project id.

jagan7157, 2018-02-01:
sid....same error getting
//---------------------------------
[Unit]
Description=Google Assistant-Voice
Wants=network-online.target
After=network-online.target

[Service]
Environment=VIRTUAL_ENV=/home/pi/env
Environment=PATH=/home/pi/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
ExecStart=/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id 'device-192519'  --device_model_id 'device-192519-20180131171828'

WorkingDirectory=/home/pi/
StandardOutput=inherit
StandardError=inherit
Restart=always
User=pi

[Install]
WantedBy=multi-user.target

shivasiddharth, 2018-02-01:
You need to have a double spaces. What you have now is:
` ExecStart=/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id 'device-192519' --device_model_id 'device-192519-20180131171828'`
It should be:
`ExecStart=/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id 'device-192519'  --device_model_id 'device-192519-20180131171828'`

jagan7157, 2018-02-01:
sid....
same problem iam getting....bad luck for me.

(env) pi@raspberrypi:~ $ sudo python GassistPi/src/main.py 
usage: main.py [-h] [--credentials OAUTH2_CREDENTIALS_FILE] --device_model_id
               DEVICE_MODEL_ID [--project_id PROJECT_ID]
main.py: error: argument --device_model_id is required

BadFlo, 2018-02-01:
Jagan, you need to start main with the arguments, or start the service, not
update the service and start "main" without arguments.

jagan7157, 2018-02-01:
sorry iam not to be understand Badflo ..can  explain  detailed with example....
     soory for this inconve
//----------------------------------------------

Jagan, you need to start main with the arguments, or start the service, not
update the service and start "main" without arguments.

shivasiddharth, 2018-02-01:
@BadFlo , Nice catch..
@jagan7157 , You need to go through this https://github.com/shivasiddharth/GassistPi#headless-autostart-on-boot-service-setup to make the assistant auto start on boot.
If you want to run it manually then open a new terminal and execute what you have after `ExecStart=`  line in systemd file.

shivasiddharth, 2018-02-20:
Closing this for inactivity.

closed 2018-02-20 by shivasiddharth