shivasiddharth/GassistPi #177: Spotify??

adas4190, 2018-02-02:
Hi, is there any chance to have it working like in real ghome? 
Thank You, you rock mate!!!!!

shivasiddharth, 2018-02-02:
The python packages for spotify has not been updated for quite sometime. There are some issues that needs to be fixed. Once the author releases the fix for the package, i will roll out the feature here.

Migushthe2nd, 2018-02-08:
The one thing I'm missing 
Sad

Migushthe2nd, 2018-02-08:
Who's the developer of those Python packages btw?

adas4190, 2018-02-08:
I use it all the time on my google homes. It would be amazing to have it on pi.

t1m0thyj, 2018-02-18:
@shivasiddharth I know that the [pyspotify](https://pypi.python.org/pypi/pyspotify/) library has not been updated in a few years. Would it be possible to use [Spotipy](https://github.com/plamere/spotipy) instead, which I think is still actively developed? I'm curious which Python packages for Spotify you were talking about.

shivasiddharth, 2018-02-18:
@t1m0thyj , I was referring to the [Spotipy](https://github.com/plamere/spotipy) only. There is this outstanding issue https://github.com/plamere/spotipy/issues/211 which has not been fixed. As it is, we can make searches and even get track/artist information, but when we try to play the tracks,  it throws out errors.

shivasiddharth, 2018-03-12:
Got the spotify working, but as it is a web based player, its sending the pi CPU utilization sky rocketing. need to take a roundabout approach.

adas4190, 2018-03-12:
Good to hear. It looks like it is nativly suported in 0.4.3 sdk. But only
when you play music supported speaker.

parthsuthar, 2018-03-28:
Any luck with this ? There are a bunch of Spotify python and Http clients available. Can gAssist Pi control Mopidy ?

R3vpanda, 2018-04-30:
@shivasiddharth  Would it be possible to have the spotify(browser) optional i am running on a 3b+ and would like to see how it works for me
Also if there is anything i can do to get Spotify going for us all let me know!

shivasiddharth, 2018-06-03:
DRM content no longer work on Pi, even when using the browser plugin to mimic other browsers.. So far its a dead end.

shivasiddharth, 2018-06-04:
Please check this commit...https://github.com/shivasiddharth/GassistPi/commit/aaaf879bd89e87f56694d032ec171a9b83fde609

If interested, try the spotify branch, after some testing, will merge it with the main branch. You need to register for a spotify developer account and add the credentials in the actions.py script. Since it uses youtube to fetch tracks, you need to add the YouTube API credentials as well. 

Right now, you can play only the user's playlists. Will add more features in the future, based on the user requests. 
Cheers..

PR #397 (2018-06-16): Spotify playback
Play tracks from your spotify playlist. Fixes #177
closed 2018-06-16 by shivasiddharth