shivasiddharth/GassistPi #178: Youtube streaming autoplay

cnoorman, 2018-02-03:
In autoplay Youtube streaming mode only one video is played. The second url in the list does'nt contain the 11 character long youtube video id. This result in phyton error in line 51 of backend_shared.py in extract_video_id 
Also the search string in that mode still contains the word AUTOPLAY (only affecting the search result).

shivasiddharth, 2018-02-03:
Please attach the log file. Log file can be found in the /tmp directory.
If you used the command as given in the example syntax, the autoplay word should be programatically removed.

cnoorman, 2018-02-04:
[dump.txt](https://github.com/shivasiddharth/GassistPi/files/1692650/dump.txt)
[GassistPi.log](https://github.com/shivasiddharth/GassistPi/files/1692651/GassistPi.log)
Enclosed log file and python traceback.

You are right about autoplay word removal. I was using the wrong order: 'stream autoplay' iso 'autoplay stream'

cnoorman, 2018-02-04:
[GassistPi.log](https://github.com/shivasiddharth/GassistPi/files/1692766/GassistPi.log)
Enclosed you find log file.

A few Import errors occur:
ImportError: No module named 'oauth2client.locked_file'
ImportError: No module named 'google.appengine'
Not all libraries up to date?

However I noticed that the webContent that is returned in line 753 of actions.py contains Dutch language. Consequently the string "Up next" is not in there. When I translate that into the Dutch "Volgende" than everything works fine.

Altough I have changed the Google/Youtube language settings into 'en' this has not resulted into the intended response. 

Issue can be closed as far as I'm concerned.
Thanks!

closed 2018-02-04 by shivasiddharth