shivasiddharth/GassistPi #179: setting assistant voice to en-GB

mwhitacre, 2018-02-03:
what are the steps to changing assistant voice to en-GB (British English) accent?

Enzoinventor, 2018-02-03:
open the folder GassistPi/src and open the file actions.py file and edit line 108.

mwhitacre, 2018-02-04:
i changed actions 108 and pushbutton 613 to en-GB but assistant still has en-US accent.

mwhitacre, 2018-02-04:
my raspberry pi was not discovered by my phones Google Home app

BadFlo, 2018-02-04:
And it will not. You can find it on your phone's assistant app in the settings.

shivasiddharth, 2018-02-05:
@mwhitacre , there are some constraints for the Pi to appear on the phone app. You must be running Android 7/N and up and also your app should be updated to the latest version.

marcominerva, 2018-02-06:
Hi! I have installed GassistPi on my Raspberry Pi and everything works correctly, however I can't see the device in the Google Home app (installed right now, so it is the last available version). I have an Android P10 Lite with Android 7. Do I need to configure something in order to make the device discoverable?

Mierola, 2018-02-06:
On my device the settings appeared for a brief moment in the "regular" Google app (not assistant or home). Nothing in the home app, and the assistant app isn´t working on my phone.

I haven´t been able to get it back after that, even if I switch language on the phone to have the same settings as when it appeared. Lucky enough I got enough time to setup all the settings so all the personal features which require that works.

shivasiddharth, 2018-03-01:
Closing this for inactivity

closed 2018-03-01 by shivasiddharth

Itega, 2018-03-20:
Hi, I installed GassistPi and it's working smoothly in english, but I'd like to use it in French. I change the configuration both in actions.py and my Assistant app but it's still listening/speaking in english. Do you have any idea about this ?

aandroide, 2018-03-20:
do you use hotword or pushbutton?

Itega, 2018-03-20:
I'm using hotword

aandroide, 2018-03-20:
opening the file gassist-installer-pi3.sh
what version of sdk do you have?

Itega, 2018-03-20:
i'm on 0.4.3

aandroide, 2018-03-20:
perfect I just wanted to take away the doubt that I was not using an old sdk version.
have you entered your device id in /home/pi/GassistPi/systemd/gassistpi-ok-google.service?

Itega, 2018-03-20:
yep I did it

aandroide, 2018-03-20:
let's try,
waste the services for a moment with:
sudo systemctl stop gassistpi-ok-google.service
sudo systemctl stop gassistpi-push-button.service
and then run: / home / pi / env / bin / python -u /home/pi/GassistPi/src/main.py --device_model_id 'replace this with the model id'
when you give the command you get the url with 200 final?

Itega, 2018-03-20:
[image]
This is what I get when I run the command

aandroide, 2018-03-20:
try adding your project id and see if so see the url mentioned above and if it works try changing the language from the assistant application now.

then run:
/ home / pi / env / bin / python -u / home / pi / GassistPi / src / maipy --project_id 'id of your project' --device_model_id 'your device model'

Itega, 2018-03-20:
Thank you it's working now !

aandroide, 2018-03-20:
thanks to you I've just discovered a bug