shivasiddharth/GassistPi #180: Does pi zero w work headless?

jabulonboaz, 2018-02-03:
Does pi zero w work in headless mode with this build or only via the button?

BadFlo, 2018-02-04:
Hi,
As written in the first few lines, RasPi0 si not powerful enough to run full Google assistant. Zero W is exactly the same hardware, so you should run button mode only. But you can try to run full mode to be sure anyways.

jabulonboaz, 2018-02-04:
How do you "run full mode" in pi zero w please?

jabulonboaz, 2018-02-04:
Also i am running this google assistant pi installation process on raspbian Jessie light (no desktop) and i got a python related error message that it could not find a .py file (cant recall exactly) then i installed python 3 as per respeaker instructions and now trying the instal again. I think raspbian jessie light is missing python?

shivasiddharth, 2018-02-04:
Pi Zero W is based on Armv6 architecture, which does not support Ok-Google wakeword. That is the reason  throughout the Readme i have clearly distinguished Pi Zero W and Pi 3 commands.
The Lite versions do not come with all the requisite python packages. I have written the installer script for Desktop OS and I have clearly mentioned the need to use the Desktop OS in the Readme in Block Letters.

closed 2018-02-04 by shivasiddharth