shivasiddharth/GassistPi #181: Service crashs on start

mirohero, 2018-02-04:
Hey,
since the last update the gassistPi crashs right after the start music is played.
I tried the update script and a fresh installation. Neither work. 
If you tell me how I can post a log.

Thank you for the great project. specially the nodeMCU addition.

Pancer1, 2018-02-04:
Check if You have your YouTube Developer key in actions.py . It solved my problems with crushing app.

mirohero, 2018-02-04:
Thank you for your reply. This doesn't solve the problem.

shivasiddharth, 2018-02-04:
Please post the output of `sudo systemctl status gassistpi-ok-google.service`
The log file can be found in /tmp directory.

mirohero, 2018-02-04:
It works now, after like 8 reboots. The service finally started successfully.

pi@raspberrypi:~ $ sudo systemctl status gassistpi-ok-google.service 
● gassistpi-ok-google.service - Google Assistant-Voice
   Loaded: loaded (/lib/systemd/system/gassistpi-ok-google.service; enabled; ven
   Active: active (running) since Sun 2018-02-04 16:47:04 UTC; 2min 19s ago
 Main PID: 468 (python)
   CGroup: /system.slice/gassistpi-ok-google.service
           └─468 /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --pro

Feb 04 16:47:04 raspberrypi systemd[1]: Started Google Assistant-Voice.
Feb 04 16:48:13 raspberrypi python[468]: device_model_id: saved-model-id
Feb 04 16:48:13 raspberrypi python[468]: device_id: 4CE5E5D0373BF0AFDBCB67FBFC4F
Feb 04 16:48:14 raspberrypi python[468]: https://embeddedassistant.googleapis.co
Feb 04 16:48:14 raspberrypi python[468]: ON_MUTED_CHANGED:
Feb 04 16:48:14 raspberrypi python[468]:   {'is_muted': False}
Feb 04 16:48:14 raspberrypi python[468]: ON_START_FINISHED

`2018-02-04 16:48:45,453 DEBUG google.auth.transport.requests Making request: POST https://accounts.google.com/o/oauth2/token
2018-02-04 16:48:45,460 DEBUG urllib3.connectionpool Starting new HTTPS connection (1): accounts.google.com
2018-02-04 16:48:45,670 DEBUG urllib3.connectionpool https://accounts.google.com:443 "POST /o/oauth2/token HTTP/1.1" 200 None
2018-02-04 16:48:45,697 INFO root Connecting to embeddedassistant.googleapis.com
2018-02-04 16:48:46,773 ERROR __main__ Error opening RawStream: Device unavailable [PaErrorCode -9985]
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/pushbutton.py", line 818, in <module>
    main()
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/env/lib/python3.5/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/GassistPi/src/pushbutton.py", line 709, in main
    flush_size=audio_flush_size
  File "/home/pi/env/lib/python3.5/site-packages/googlesamples/assistant/grpc/audio_helpers.py", line 187, in __init__
    blocksize=int(block_size/2),  # blocksize is in number of frames.
  File "/home/pi/env/lib/python3.5/site-packages/sounddevice.py", line 1256, in __init__
    **_remove_self(locals()))
  File "/home/pi/env/lib/python3.5/site-packages/sounddevice.py", line 772, in __init__
    'Error opening {0}'.format(self.__class__.__name__))
  File "/home/pi/env/lib/python3.5/site-packages/sounddevice.py", line 2563, in _check
    raise PortAudioError(errormsg, err)
sounddevice.PortAudioError: Error opening RawStream: Device unavailable [PaErrorCode -9985]
2018-02-04 16:48:54,415 DEBUG google.auth.transport.requests Making request: POST https://accounts.google.com/o/oauth2/token
2018-02-04 16:48:54,421 DEBUG urllib3.connectionpool Starting new HTTPS connection (1): accounts.google.com
2018-02-04 16:48:54,627 DEBUG urllib3.connectionpool https://accounts.google.com:443 "POST /o/oauth2/token HTTP/1.1" 200 None
2018-02-04 16:48:54,660 INFO root Connecting to embeddedassistant.googleapis.com
2018-02-04 16:48:55,725 ERROR __main__ Error opening RawStream: Device unavailable [PaErrorCode -9985]
Traceback (most recent call last):
`

mirohero, 2018-02-04:
I works now. But if I reboot my pi it restarts the service like 8 times in the beginning.

shivasiddharth, 2018-02-04:
You seem to have enabled the pushbutton.py gRPC API google assistant as well. Your Ok-Goolge assistant is working fine without issues. Disable the pushbutton.py assistant using:
```
sudo systemctl stop gassistpi-push-button.service
sudo systemctl disable gassistpi-push-button.service
```

mirohero, 2018-02-04:
It works now. Thank you.

closed 2018-02-04 by shivasiddharth